autowarefoundation/autoware_universe #1: Merge packages in Tier IV's AutowareArchitectureProposal repository

mitsudome-r, 2021-11-30:
Merge packages in https://github.com/tier4/AutowareArchitectureProposal.iv

kenji-miyake, 2021-11-30:
This is the directory structure that @mitsudome-r has discussed and agreed with AWF core members in [this meeting](https://gitlab.com/autowarefoundation/autoware-foundation/-/issues/184#directory-structure-of-universe).
[image]

If it gets messy, we can consider a new structure.
However basically, we should refactor, write design documents with rationales, and move core packages to https://github.com/autowarefoundation/autoware.core.

kenji-miyake, 2021-11-30:
Although there are DCO errors, we'll ignore them for `tier4/proposal` branch.
https://github.com/autowarefoundation/autoware.universe/pull/3#issuecomment-982235871

When we merge it to `main` branch, @mitsudome-r will sign the squashed commits.

kenji-miyake, 2021-11-30:
The commit hash we started porting is this.
https://github.com/tier4/AutowareArchitectureProposal.iv/commit/c23d3e1ee46e0dfd9db194949b6c599b477f814f

kenji-miyake, 2021-12-27:
@tkimura4 @taikitanaka3 @1222-takeshi Could you explain the remaining tasks, please? Nothing?

taikitanaka3, 2022-01-12:
@kenji-miyake 
I think this issue can be closed.

kenji-miyake, 2022-01-12:
Cool! After @mitsudome-r checks it, let's close!.

kenji-miyake, 2022-02-01:
I believe we can close this.

closed 2022-02-01 by kenji-miyake