autowarefoundation/autoware_universe #21: Remove unnecessary messages

1222-takeshi, 2021-11-30:
Check the dependency and remove unnecessary messages that are replaced by autoware_auto_msgs from tier4_msgs.

kenji-miyake, 2021-12-04:
Also, the naming should be refined. For example, to `t4_planning_msgs`, `tier4_*_msgs` or so.

1222-takeshi, 2021-12-09:
message type that may still be used.
To be removed.

- [ ] autoware_perception_msgs/object_recognition/DynamicObject.msg
- [ ] autoware_perception_msgs/object_recognition/DynamicObjectArray.msg
- [ ] autoware_perception_msgs/object_recognition/PredictedPath.msg
- [ ] autoware_perception_msgs/object_recognition/Semantic.msg
- [ ] autoware_perception_msgs/object_recognition/Shape.msg
- [ ] autoware_perception_msgs/object_recognition/State.msg
- [ ] autoware_planning_msgs/LaneChangeCommand.msg (maybe unnecessary)
- [ ] autoware_planning_msgs/LaneChangeStatus.msg (maybe unnecessary)
- [ ] autoware_vehicle_msgs/BatteryStatus.msg

closed 2021-12-17 by 1222-takeshi