autowarefoundation/autoware_universe #88: Some of the reference articles are written by Japanese

taikitanaka3, 2021-12-02:
Some packages of readme or comment has reference which is written by Japanese.

kenji-miyake, 2021-12-27:
@taikitanaka3 Could you please list the Japanese documents?

kenji-miyake, 2022-03-24:
@taikitanaka3 Could you update the status?

taikitanaka3, 2022-03-25:
@kenji-miyake 
This issue is to change Japanese documents to English as much as possible and currently most of documents are written in English so I think I can close this.

closed 2022-03-25 by taikitanaka3

08beeqtufail-sudo, 2022-04-27:
Hello everyone,
While going through the documentation found at 
[Autoware universe documentation](https://autowarefoundation.github.io/autoware.universe/latest/planning/behavior_velocity_planner/intersection-design/) 
I found out that the images have all the explanation in Japanese. Can someone provide an English version or just a brief explanation of images in English . Thank you for great work.