autowarefoundation/autoware_universe #119: Fix subscriber qos for route

wep21, 2021-12-06:
https://github.com/tier4/autoware.iv/pull/2388

kenji-miyake, 2021-12-06:
https://github.com/tier4/AutowareArchitectureProposal.iv/pull/763

wep21, 2021-12-07:
https://github.com/autowarefoundation/autoware.universe/pull/137

1222-takeshi, 2021-12-08:
@wep21 
I think this Issue has been resolved.
Could you close this Issue ?

wep21, 2021-12-08:
@1222-takeshi There are some other nodes which uses a defaut qos for route.
```
❯ ag "create_subscription<autoware_auto_planning_msgs::msg::HADMapRoute>\(" ./ -A 3
system/autoware_state_monitor/src/autoware_state_monitor_node/autoware_state_monitor_node.cpp
432:  sub_route_ = this->create_subscription<autoware_auto_planning_msgs::msg::HADMapRoute>(
433-    "input/route", rclcpp::QoS{1}.transient_local(),
434-    std::bind(&AutowareStateMonitorNode::onRoute, this, _1), subscriber_option);
435-  sub_odom_ = this->create_subscription<nav_msgs::msg::Odometry>(

perception/traffic_light_map_based_detector/src/node.cpp
105:  route_sub_ = create_subscription<autoware_auto_planning_msgs::msg::HADMapRoute>(
106-    "~/input/route", 1, std::bind(&MapBasedDetector::routeCallback, this, _1));
107-
108-  // publishers

planning/scenario_selector/src/scenario_selector_node/scenario_selector_node.cpp
348:  sub_route_ = this->create_subscription<autoware_auto_planning_msgs::msg::HADMapRoute>(
349-    "input/route", rclcpp::QoS{1},
350-    std::bind(&ScenarioSelectorNode::onRoute, this, std::placeholders::_1));
351-  sub_odom_ = this->create_subscription<nav_msgs::msg::Odometry>(

planning/mission_planner/src/goal_pose_visualizer/goal_pose_visualizer.cpp
22:  sub_route_ = create_subscription<autoware_auto_planning_msgs::msg::HADMapRoute>(
23-    "input/route", rclcpp::QoS{1},
24-    std::bind(&GoalPoseVisualizer::echoBackRouteCallback, this, std::placeholders::_1));
25-  pub_goal_pose_ = create_publisher<geometry_msgs::msg::PoseStamped>(
```

1222-takeshi, 2021-12-09:
@wep21 
I created this PR. https://github.com/autowarefoundation/autoware.universe/pull/149
Please check this.

PR #149 (2022-01-13): fix: use transient_local qos for route
## Related Issue(required)
Resolves #119 
## Description(required)
fix subscriber qos for route
## Review Procedure(required)
Make sure that the qos of all route subscribers is set to transient_local.

``` sh
ag "create_subscription<autoware_auto_planning_msgs::msg::HADMapRoute>\(" ./ -A 3
```
## Related PR(optional)

## Pre-Review Checklist for the PR Author

**PR Author should check the checkboxes below when creating the PR.**

- [ ] Read [commit-guidelines][commit-guidelines]
- [ ] Assign PR to reviewer

If you are adding new package following items ar [... cut]
closed 2022-01-13 by kenji-miyake