autowarefoundation/autoware_universe #141: Rename existing packages named `autoware_*` to different names

kenji-miyake, 2021-12-07:
To avoid confusion between Core/Common's prefix.
https://github.com/autowarefoundation/autoware_common/issues/3#issuecomment-987950199

tkimura4, 2021-12-10:
Please comment if there is something better.
@kenji-miyake 
@mitsudome-r 

Suggestion.
---
autoware_api_utils -> tier4_api_utils
autoware_state_monitor -> driving_state_monitor
autoware_error_monitor -> system_error_monitor
autoware_utils -> tier4_autoware_utils
autoware_global_parameter_loader -> global_parameter_loader
autoware_iv_auto_msgs_converter -> tier4_auto_msgs_converter
autoware_joy_controller -> tier4_autoware_joy_controller
autoware_web_controller -> tier4_autoware_web_contoller

autoware_datetime_rviz_plugin -> TBD.
autoware_localization_rviz_plugin -> TBD.
autoware_state_rviz_plugin -> TBD.
autoware_vehicle_rviz_plugin -> TBD.

kenji-miyake, 2021-12-10:
@tkimura4 Thank you!

> autoware_state_monitor -> driving_state_monitor

I'm considering if it should be `ad_service_state_monitor`. :thinking: 

@isamu-takagi Do you have any idea?

> autoware_joy_controller -> tier4_autoware_joy_controller
> autoware_web_controller -> tier4_autoware_web_contoller

I feel just `joy_controller` and `web_controller` are acceptable.

> autoware_version -> tier4_autoware_version

Can this be removed?
I guess @isamu-takagi and @yamacir-kit can judge it.
If it cannot yet, I think changing the package name will break the API.

> autoware_rosbag_recorder -> tier4_autoware_rosbag_recorder

Since no one uses it anymore, I think we can remove it.

> autoware_*_rviz_plugins

Hmm, `tier4_localization_plugins` sounds good to me.
Or putting everything into one package `tier4_autoware_rviz_plugins` is better?

kenji-miyake, 2021-12-10:
@yukkysaito Please comment if you have any opinions. :pray:

tkimura4, 2021-12-10:
autoware_api_utils -> tier4_api_utils
autoware_debug_tools -> tier4_autoware_debug_tools
autoware_error_monitor -> system_error_monitor
autoware_state_monitor -> ad_service_state_monitor
autoware_utils -> tier4_autoware_utils
autoware_global_parameter_loader -> global_parameter_loader
autoware_iv_auto_msgs_converter -> tier4_auto_msgs_converter
autoware_joy_controller -> joy_controller
autoware_web_controller -> web_contoller
autoware_version -> tier4_autoware_version? (TBD.)
autoware_rosbag_recorder -> remove

autoware_*_rviz_plugin -> tier4_*_rviz_plugin (?)
autoware_datetime_rviz_plugin, autoware_localization_rviz_plugin, autoware_perception_rviz_plugin, autoware_planning_rviz_plugin, autoware_state_rviz_plugin, autoware_vehicle_rviz_plugin,

1222-takeshi, 2021-12-10:
@tkimura4 @kenji-miyake 
What do you think about autoware_auto_perception_rviz_plugin?

yamacir-kit, 2021-12-10:
@kenji-miyake 
Since scenario_simulator_v2 does not yet depend on autoware_version, renaming autoware_version will not cause any problems.

isamu-takagi, 2021-12-10:
>autoware_state_monitor -> driving_state_monitor
>I'm considering if it should be ad_service_state_monitor. thinking
>@isamu-takagi Do you have any idea?

@tkimura4 @kenji-miyake 
autoware_state_monitor will merged into autoware_iv_external_api_adaptor,
so I think it is better to rename temporary name. 
Then, external_api_adaptor shoud be ad_service_state_monitor or adapi_adaptor or something.

>autoware_version -> tier4_autoware_version
>Can this be removed?

@tkimura4 @yamacir-kit 
yes, it can be removed if simulator does not use.

tkimura4, 2021-12-13:
update:

autoware_api_utils -> tier4_api_utils
autoware_debug_tools -> tier4_autoware_debug_tools
autoware_error_monitor -> system_error_monitor
autoware_state_monitor -> ad_service_state_monitor
autoware_utils -> tier4_autoware_utils
autoware_global_parameter_loader -> global_parameter_loader
autoware_iv_auto_msgs_converter -> tier4_auto_msgs_converter
autoware_joy_controller -> joy_controller
autoware_web_controller -> web_contoller
autoware_version -> remove
autoware_rosbag_recorder -> remove

autoware_*_rviz_plugin -> tier4_*_rviz_plugin
autoware_datetime_rviz_plugin, autoware_localization_rviz_plugin, autoware_perception_rviz_plugin, autoware_planning_rviz_plugin, autoware_state_rviz_plugin, autoware_vehicle_rviz_plugin,

autoware_auto_perception_rviz_plugin -> No change

1222-takeshi, 2021-12-17:
@tkimura4 
Can I close this Issue ?

tkimura4, 2021-12-17:
@1222-takeshi 
Yes. Thanks!

closed 2021-12-17 by tkimura4