autowarefoundation/autoware_universe #143: Consider using global_parameter_loader

kenji-miyake, 2021-12-07:
global parameters

With our current plan, there will be multiple `default_common.param.yaml`, but it might increase the maintenance cost, for example when the common parameter gets a new variable.

- https://github.com/autowarefoundation/autoware.universe/blob/c2e79c3a56956f867030b2cbc9675e2e7c7c1339/planning/external_velocity_limit_selector/config/default_common.param.yaml
- https://github.com/autowarefoundation/autoware.universe/blob/c2e79c3a56956f867030b2cbc9675e2e7c7c1339/planning/motion_velocity_smoother/config/default_common.param.yaml

To avoid this, we can consider `global_parameter_loader`.

https://github.com/autowarefoundation/autoware.universe/blob/c2e79c3a56956f867030b2cbc9675e2e7c7c1339/planning/behavior_velocity_planner/launch/behavior_velocity_planner.launch.xml#L3

https://github.com/autowarefoundation/autoware.universe/blob/c2e79c3a56956f867030b2cbc9675e2e7c7c1339/common/autoware_global_parameter_loader

### Without global_parameter_loader

launch.xml:

```xml
// planning.launch.xml
<launch>
  <include file="behavior_planner.launch.xml">
    <arg name="common_param_path" value="{path_to_file}" />
    <arg name="vehicle_param_path" value="{path_to_file}" />
  </include>
</launch>

// behavior_planner.launch.xml
<launch>
  // Don't consider the default values here because you can't use them when you debug in the real environment.
  <arg name="common_param_path" />
  <arg name="vehicle_param_path" />

  <node ...>
  </node>
</launch>
```

How to launch:

```sh
// autoware.launch.xml
ros2 launch autoware_launch autoware.launch.xml vehicle_model:={vehicle_name}

// behavior_planner.launch.xml
ros2 launch planning_launch behavior_planner.launch.xml common_param_path:={path_to_file} vehicle_info_param_path:={path_to_file}
```

### With global_parameter_loader

launch.xml:

```xml
// planning.launch.xml
<launch>
  <include file="behavior_planner.launch.xml">
    // No arg
  </include>
</launch>

// behavior_planner.launch.xml
<launch>
  <include file="$(find-pkg-share autoware_global_parameter_loader)/launch/global_params.launch.py"/>

  <node ...>
  </node>
</launch>
```

How to launch:

```sh
// autoware.launch.xml
ros2 launch autoware_launch autoware.launch.xml vehicle_model:={vehicle_name}

// behavior_planner.launch.xml
ros2 launch planning_launch behavior_planner.launch.xml vehicle_model:={vehicle_name}
```

esteve, 2022-05-26:
@kenji-miyake this is how we solved this issue in Autoware.Auto: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/merge_requests/978 The solution was to move the vehicle characteristics (which are common to many packages and were duplicated everywhere) to a separate params file and then load it from launch files. The main difference between Autoware and Autoware.Auto is that we are using XML files to launch packages, but I think the Python scripts are more flexible.

esteve, 2022-06-15:
@kenji-miyake are you actively working on this? I'd be happy to help :slightly_smiling_face:

kenji-miyake, 2022-06-15:
@esteve Thank you for your confirmation.
Actually, no. I'm not working on this now.

If you can help with this, I think it's good to start by collecting developers' opinions in https://github.com/orgs/autowarefoundation/discussions/416.
It's because there are no big technical issues and we only have to decide the design considering trade-offs of possible solutions.

Or I think it's also good to submit a draft PR so that everyone can have a more concrete image, and discuss the design there.

yukkysaito, 2022-10-18:
I will try to work on it this month.

yukkysaito, 2022-11-30:
I'm working in [this PR](https://github.com/autowarefoundation/autoware_launch/pull/116)

mitsudome-r, 2026-03-24:
I'm closing the issue that was inactive for a long time. Feel free to reopen if the issue is still relevant.