autowarefoundation/autoware_universe #204: Support Docker development environment

kenji-miyake, 2021-12-20:
To provide an easy-to-develop environment, we should support Docker.

The base of Dockerfile is introduced here.
https://github.com/tier4/AutowareArchitectureProposal.proj/tree/main/docker

Also, Autoware.Auto has Dockerfile for `ade`.
https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/tools/ade_image/Dockerfile

kenji-miyake, 2021-12-20:
To be able to build not only `autoware.universe` but also the entire workspace including drivers, tools, and etc., I think it's better to put Dockerfile into the meta-repository.

So I'll wait for we'll set up https://github.com/autowarefoundation/autoware.

For example, we need CUDA, TensorRT, and PyTorch for Universe for now, even though they might be replaced by TVM in the future.
https://github.com/tier4/AutowareArchitectureProposal.proj/tree/main/ansible/roles

kenji-miyake, 2021-12-21:
This Docker image can be used in the CI of `autoware.universe`. Currently, it uses `ros:galactic`, but since it doesn't have CUDA, the Perception packages are not built.

kenji-miyake, 2022-01-13:
Implemented the first version in https://github.com/autowarefoundation/autoware/pull/14.

IshitaTakeshi, 2022-01-13:
I'd suggest some features to add:
* The execution procedure. How to launch autoware and play rosbag
* Mount files to the host environment. It should be easier to modify files on the host environment rather than on a container

kenji-miyake, 2022-02-02:
> The execution procedure. How to launch autoware and play rosbag

Will be covered by https://github.com/autowarefoundation/autoware-documentation/issues/8.

> Mount files to the host environment. It should be easier to modify files on the host environment rather than on a container

I wrote a document in this PR.
https://github.com/autowarefoundation/autoware/pull/14

closed 2022-02-02 by kenji-miyake