autowarefoundation/autoware_universe #221: Diagnostics with no corresponding node

0x126, 2021-12-24:
There are no nodes that output these diagnostics in this repository or in any other public repository.
Don't these need to be removed or commented out?

https://github.com/autowarefoundation/autoware.universe/blob/tier4/proposal/system/system_error_monitor/config/diagnostic_aggregator/control.param.yaml#L69-L81

https://github.com/autowarefoundation/autoware.universe/blob/tier4/proposal/system/system_error_monitor/config/diagnostic_aggregator/map.param.yaml#L17-L25

kenji-miyake, 2021-12-24:
Generally, regarding diagnostic_aggregator configs, I think no need to remove them because we can adjust whether to watch them in `autoware_error_monitor.param.yaml`.

But since we don't use `map_version` for now, we can remove it now and re-consider the name when it's added.

0x126, 2022-01-18:
Thank you for your comment, I created a PR.

closed 2022-02-01 by 0x126