autowarefoundation/autoware_universe #248: Merge packages from autoware repo

angry-crab, 2022-01-13:
This issue aims to merge packages from autoware repo in gitlab. Prefix 'auto' will be added to packages from gitlab to distinguish with those in `universe`. The current plan is to add most packages except for the ones under `src/tools`. Please feel free to make any comment or suggestion.

kenji-miyake, 2022-01-13:
@angry-crab Thank you for opening this issue! :smile: 

How will you deal with the packages that are already ported from `Autoware.Auto` like `trajectory_follower_nodes`?
https://github.com/autowarefoundation/autoware.universe/blob/3e27b12760dfb6fad8b686ea82e8f4a768297d4d/control/trajectory_follower_nodes/package.xml

Will you prefix them as well?
I think it's better to discuss with @TakaHoribe @maxime-clem @taikitanaka3.

Also, when will you apply the new formatter?
Currently, packages from Autoware.Auto are ignored [here](https://github.com/autowarefoundation/autoware.universe/blob/3e27b12760dfb6fad8b686ea82e8f4a768297d4d/.pre-commit-config.yaml#L92) in case we'll sync the code after Cargo ODD cleanup.
My suggestion is after your work is done we'll format all packages.

maxime-clem, 2022-01-13:
I thought the code from Autoware.auto would become https://github.com/autowarefoundation/autoware.core. Is this wrong ?

kenji-miyake, 2022-01-13:
@maxime-clem Yes, I believe it's wrong. We need refactoring even it's from Autoware.Auto, they contain some code developed in a prototyping level.

TIER IV will propose the merge criteria for Core/Universe and AWF members will discuss it.
https://github.com/autowarefoundation/autoware-documentation/issues/2

The plan was proposed here.
https://gitlab.com/autowarefoundation/autoware-foundation/-/issues/189

In order to keep the quality, we shouldn't merge to `autoware.core` before we can agree with the criteria.

xmfcx, 2022-01-13:
> @angry-crab Thank you for opening this issue! smile
>
> How will you deal with the packages that are already ported from `Autoware.Auto` like `trajectory_follower_nodes`? [...]

I think we need to explain/show what currently exists in the universe right now since there was no merge criteria with what is already in the universe right now.

What are current capabilities of the universe?

We don't need to merge all of the Autoware.Auto to the universe. We can always take packages/code snippets from the Gitlab repository as needed.

We should first understand what exists in the universe right now and explain it to the Autoware.Auto contributors.

> Also, when will you apply the new formatter? Currently, packages from Autoware.Auto are ignored [...]

`exclude: .svg|control/trajectory_follower|control/trajectory_follower_nodes|common/autoware_auto_cmake|common/autoware_auto_common|common/autoware_auto_geometry|common/autoware_auto_tf2|common/autoware_testing|common/fake_test_node|common/had_map_utils|common/motion_common|common/motion_testing|common/time_utils|common/vehicle_constants_manager|common/autoware_auto_perception_rviz_plugin|common/osqp_interface|simulator/simple_planning_simulator|planning/freespace_planner|planning/astar_search|planning/costmap_generator`

@kenji-miyake  How did you decide to add these specific packages?

Why did you add these to the universe?

angry-crab, 2022-01-13:
@kenji-miyake Thanks for the comments. I think we could ignore the packages which are already ported. 
As for the new formatter, do you mean code style? If so, we should leave it for the refactoring step.

kenji-miyake, 2022-01-13:
@xmfcx Thank you.

> I think we need to explain/show what currently exists in the universe right now since there was no merge criteria with what is already in the universe right now.

Yes, I'll consider and prepare documents. Please wait for a while. :bow: 
For now, since I've set up the CI, it is the minimum requirement, I think.

> What are current capabilities of the universe?

Did you mean a kind of functional coverage of Autonomous Driving?
Then full features of TIER IV's proposal were merged.

> We don't need to merge all of the Autoware.Auto to the universe. We can always take packages/code snippets from the Gitlab repository as needed.

Yes, I agree.
@angry-crab So could you list up the necessary features from Autoware.Auto?

> We should first understand what exists in the universe right now and explain it to the Autoware.Auto contributors.

I see. Then we need some tutorial documents for that, right?

> @kenji-miyake How did you decide to add these specific packages?
Why did you add these to the universe?

It's because:
- TIER IV had already ported a part of our proposal packages to Autoware.Auto.
- We thought using the reviewed code was better rather than using the original TIER IV's proposal code, so ported them back.
- Some dependent packages were necessary as well, but we couldn't spare time for making the dependency simple.

@xmfcx Is it clear for you?
@TakaHoribe @maxime-clem Am I right?

maxime-clem, 2022-01-13:
> @angry-crab So could you list up the necessary features from Autoware.Auto?

I think the packages required to run all the demos (see `autoware_demos` in .Auto) can be a good start.
 
> @xmfcx Is it clear for you? @TakaHoribe @maxime-clem Am I right?

This is my understanding as well. Unfortunately I am afraid some packages have diverged between .Auto and universe so we will need to do some checking to not lose any recent improvements.

angry-crab, 2022-01-14:
@kenji-miyake Thanks for clarification. If the universe repo has already been using a new formatter, I'll make sure pacckages from` Autoware.auto`  updated as well. Otherwise, I'll leave it for now. 
As for the candidates list, I agree with @maxime-clem that we can start with packages used in `autoware_demo`.

angry-crab, 2022-01-18:
Another issue is that whether we should prefix the namespace, name of class, and name of folder. After discussing with @mitsudome-r-san, I think those should be prefixed to avoid ambiguity for development process.

kenji-miyake, 2022-03-24:
@angry-crab Could you write the latest status?

angry-crab, 2022-03-25:
@kenji-miyake After discussing @xmfcx  and @mitsudome-r , we've decided to postponed this merge.

angry-crab, 2022-03-25:
Originally, the intention of this merge is to make `avp` demo available for `universe`. However, since `svl` is going down. The amount of packages is too much. Therefore, if any package from `autoware.auto` is required, please feel free to include it by yourself. If there are really a lot of dependencies required, I'll reopen this issue and move them in batch.

closed 2022-03-25 by angry-crab