autowarefoundation/autoware_universe #255: Fix calcuation of angle difference

kyoichi-sugahara, 2022-01-16:
In the function of angle difference calculation, NaN value is returned when the absolute value of inner product is larger than 1.
https://github.com/autowarefoundation/autoware.universe/blob/045009c4834f9c0ae9c2a0513e6f6c12ee88e86a/map/lanelet2_extension/lib/query.cpp#L38-L45

This is because arg of atd::acos shouldn't be outside the range [-1.0, 1.0] expalined [here](https://en.cppreference.com/w/cpp/numeric/math/acos).
Numerical error happens in some cases so gurad needs to be added.

closed 2022-01-18 by taikitanaka3

kenji-miyake, 2022-01-18:
@kyoichi-sugahara Could you explain specifically in what situations the inner product will be larger than 1, please?

kyoichi-sugahara, 2022-01-18:
@kenji-miyake 
Sorry for not describing the detail.
I had same a same problem with the [link](https://stackoverflow.com/questions/52138147/why-is-acos-resulting-in-nanind-when-using-the-result-of-a-dot-product)

When `angle1` and `angle2` are the same, `vec1.dot(vec2)` expected to be 1. but in some cases, the value which is larger than 1 is calculated because of numerical error.