autowarefoundation/autoware_universe #265: some of calcSignedArcLength function need to consider path index

taikitanaka3, 2022-01-18:


kenji-miyake, 2022-01-18:
Please write some descriptions.

taikitanaka3, 2022-01-18:
currently calcSignedArclength is only consider position to the closest segment so the place where path or trajectory has it's own intersection segment then closest point is considered as target point.
[image]

taikitanaka3, 2022-02-17:
for other cases
[image]

taikitanaka3, 2022-02-17:
use case
[image]

taikitanaka3, 2022-02-17:
@takayuki5168 
Do you have any idea how to avoid segmentation fault if application side wants calculate longitudinal offset segment from last path index?
https://github.com/autowarefoundation/autoware.universe/blob/ca45e4aa3af8837fd1853f69d036ea9c1ec2d2a3/common/tier4_autoware_utils/include/tier4_autoware_utils/trajectory/trajectory.hpp#L128

taikitanaka3, 2022-04-20:
closed for stop line module has this fix

closed 2022-04-20 by taikitanaka3