autowarefoundation/autoware_universe #268: Support rolling(humble)

wep21, 2022-01-18:


wep21, 2022-01-18:
https://docs.ros.org/en/foxy/Releases/Release-Humble-Hawksbill.html

kenji-miyake, 2022-03-24:
We have to plan migration steps to Humble.

My (super rough) proposal is:
- Support rolling after Ubuntu 22.04 is released. (Around the end of April.)
  - If a change is supported by Galactic, merge to the `main` branch.
  - If not, merge to the `rolling` branch.
- Support Humble after ROS 2 Humble is released. (Around the end of May.)
  - Ideally, just renaming `rolling` to `humble`.
- Create the `galactic` branch and merge `humble` to `main`.

@xmfcx @mitsudome-r What do you think about it?

kenji-miyake, 2022-04-18:
https://github.com/autowarefoundation/autoware.universe/labels/rolling
https://github.com/autowarefoundation/autoware.universe/labels/humble

kenji-miyake, 2022-04-18:
I'll set up Ubuntu 22.04 next week and work in a branch.

kenji-miyake, 2022-04-21:
TODO: Remove workarounds after we moved to Humble.
https://github.com/autowarefoundation/autoware.universe/pull/720

kenji-miyake, 2022-05-05:
@esteve From your ROS experiences, do you have any suggestions on this?
https://github.com/autowarefoundation/autoware.universe/issues/268#issuecomment-1078430411

esteve, 2022-05-05:
@kenji-miyake in my opinion, the plan sounds good. After Humble is released, we won't be supporting Rolling, right?

kenji-miyake, 2022-05-05:
@esteve No, I believe we won't (at least proactively).
But if someone wants to add the Rolling support by using `ROS_DISTRO` macros, I think it can be acceptable.
However, it cannot be a mandatory condition for merging PRs because Rolling may collapse frequently.

kenji-miyake, 2022-07-04:
Now the `main` branch supports both `Galactic` and `Humble`.
See https://github.com/autowarefoundation/autoware/pull/2692 for further information.

closed 2022-07-04 by kenji-miyake