autowarefoundation/autoware_universe #286: Display code coverage in pull requests

KeisukeShima, 2022-01-24:
Related to <https://github.com/autowarefoundation/autoware-documentation/issues/2>, I would like to add the feature to measure and display code coverage for pull requests to improve review process.

Considerations
* About the reporter, whether to use CodeCov or to create a function to measure and report in GitHub Actions.
* How to specify labels for coverage by label, whether to automatically collect all labels or to specify the labels we want to set.

Translated with www.DeepL.com/Translator (free version)

kenji-miyake, 2022-01-24:
:+1: 
I think using CodeCov at first is efficient.

kenji-miyake, 2022-02-01:
@KeisukeShima Can this be closed?
Or will you work on `coverage by label` in this issue? If so, I think the title should be changed.

KeisukeShima, 2022-02-01:
> @KeisukeShima Can this be closed? Or will you work on `coverage by label` in this issue? If so, I think the title should be changed.

I think that getting coverage by label is also part of this issue.
If we need to change the title, how about changing it to `Display labeled code coverage in pull request`?

kenji-miyake, 2022-06-21:
I think we can close this for now. (Codecov integration is done.)
We'll reconsider this in https://github.com/autowarefoundation/autoware.core.

closed 2022-06-21 by kenji-miyake