autowarefoundation/autoware_universe #361: There are duplicate parameters and functions in bhavior_velocity_planner

kosuke55, 2022-02-08:
There are duplicate parameters and functions in bhavior_velocity_planner. It's better to common them.
- Functions
  -  calcSignedArcLength
- Parameters

mitsudome-r, 2022-04-19:
@taikitanaka3 Are there still duplicate parameters functions in the planner node?

taikitanaka3, 2022-04-19:
@mitsudome-r yes there still duplicated packages like autoware auto.utils.
but i think we wont fix in a month so.we can close issue for now and link this issue when refactoring.
@kosuke55 what do you.think？

kosuke55, 2022-04-20:
@mitsudome-r @taikitanaka3 
We haven't listed them yet because there are so many, but they're still there. (ex: behavior_velocity parameters are not commoned) 
It's OK to close this issue if this doesn't work. (or every time we find duplicates, we will list them here to help refactor.)

mitsudome-r, 2022-06-21:
@kosuke55 I will close this issue for now. Please feel free to reopen or recreate a new issue if you have time to work on this.

closed 2022-08-18 by takayuki5168