autowarefoundation/autoware_universe #372: CodeCov does not show correct coverage diffs in PR comments

KeisukeShima, 2022-02-10:
This is due to the fact that only the packages that have changed are being built and tested by CI.
I am looking for a solution.

ex. https://github.com/autowarefoundation/autoware.universe/pull/370#issuecomment-1034434086
[image]

KeisukeShima, 2022-02-10:
Do I have to merge the differences with the overall coverage before uploading them?

[image]

KeisukeShima, 2022-02-14:
> The Coverage diff in the comments shows up fine, but the list of changed files still has problems.

Now, the list of changed files works well.
https://github.com/KeisukeShima/demos/pull/2#issuecomment-1038931083

The necessary requirements are listed in the comments below.
https://github.com/KeisukeShima/demos/pull/2#issuecomment-1038969113

kenji-miyake, 2022-02-16:
Could you summarize the problem, the solution, and the evidence, in addition to the results, please? :bow:

KeisukeShima, 2022-02-17:
## Problem

CodeCov does not show correct coverage diffs in PR comments.

[image]

## Solution

Merge the coverage generated by the differential build with the full coverage results from the base branch. This tells CodeCov that there are no changes to the packages that were not built in the diff build.
When merging, you need to specify `branch_coverage=1`.

https://github.com/autowarefoundation/autoware.universe/pull/384/files#diff-0e24b6901e8528bd669e9e78acb1327c99dd72aed1affe58b818eb65899e1ce6R60

```mermaid
flowchart LR
    A[Diff coverage] --> B[Merged coverage]
    C[Total coverage] --> B
    B --> CodeCov
```

Add `codecov.yaml` to the default branch.
Include the following items in the yaml
Set `flags` and enable `carryforward`. This is to carry forward the coverage of the base branch for packages that are not built in the diff build.

```yaml
flags:
  total:
    paths:
      - ".*"
    carryforward: true
```

Upload the coverage file, including the flags, to CodeCov. This is to enable the codecov.yaml settings.

```yaml
      - name: Upload coverage to CodeCov
        if: ${{ steps.check-file-existence.outputs.exists == 'true' }}
        uses: codecov/codecov-action@v2
        with:
          files: lcov/total_coverage.info,coveragepy/.coverage
          fail_ci_if_error: false
          verbose: true
          flags: total
```

## Evidence

I confirmed that it works with my fork.
https://github.com/KeisukeShima/demos/pull/2#issuecomment-1038931083

## Result

As shown in the linked image below, the CodeCov coverage report is displayed correctly in PR.
https://github.com/KeisukeShima/demos/pull/2#issuecomment-1038931083

kenji-miyake, 2022-02-17:
@KeisukeShima Great! Thank you for your clear explanation! :smile:

KeisukeShima, 2022-03-29:
Related
https://community.codecov.com/t/carryforward-flag-problems-with-delta-testing/3388/3