autowarefoundation/autoware_universe #382: Fix errors of clang-tidy-pr-comments

kenji-miyake, 2022-02-14:
- https://github.com/autowarefoundation/autoware.universe/pull/332/files#r805538343

kenji-miyake, 2022-02-15:
Sometimes it fails to check out?
https://github.com/autowarefoundation/autoware.universe/runs/5194193198?check_suite_focus=true

kenji-miyake, 2022-03-03:
Not errors but a bit noisy.
https://github.com/autowarefoundation/autoware.universe/pull/445#discussion_r818294628
https://github.com/autowarefoundation/autoware.universe/pull/445#discussion_r818294635

esteve, 2022-06-15:
@kenji-miyake is this issue still valid? I haven't seen this test fail on the CI lately

kenji-miyake, 2022-06-15:
@esteve I guess it's still valid.

> I haven't seen this test fail on the CI lately

It's because the workflow is disabled now.

[image]

The reason why I disabled it was it's noisy.

[image]

kenji-miyake, 2022-06-15:
I think there are two ways to test this. (Except for the [checkout issue](https://github.com/autowarefoundation/autoware.universe/issues/382#issuecomment-1039817486).)
1. Re-enable the workflow.
2. Download the artifact of the `build-and-test-differential` workflow, and test it locally.
[image]

kenji-miyake, 2022-06-15:
I re-enabled the workflow. If it's still noisy and contains a lot of errors, I'll disable it again. :disappointed:

kenji-miyake, 2022-06-15:
@esteve What do you think about this result?
https://github.com/autowarefoundation/autoware.universe/pull/997

kenji-miyake, 2022-06-15:
Regarding these proposals, they should be done at once.
[image]

```yaml
  - DiagnosticName:  modernize-pass-by-value
    DiagnosticMessage:
      Message:         'pass by value and use std::move'
      FilePath:        '/__w/autoware.universe/autoware.universe/planning/behavior_path_planner/src/scene_module/scene_module_bt_node_interface.cpp'
      FileOffset:      1309
      Replacements:
        - FilePath:        '/__w/autoware.universe/autoware.universe/planning/behavior_path_planner/include/behavior_path_planner/scene_module/scene_module_bt_node_interface.hpp'
          Offset:          1453
          Length:          42
          ReplacementText: 'std::shared_ptr<SceneModuleStatus> '
        - FilePath:        '/__w/autoware.universe/autoware.universe/planning/behavior_path_planner/src/scene_module/scene_module_bt_node_interface.cpp'
          Offset:          1309
          Length:          42
          ReplacementText: 'std::shared_ptr<SceneModuleStatus> '
        - FilePath:        '/__w/autoware.universe/autoware.universe/planning/behavior_path_planner/src/scene_module/scene_module_bt_node_interface.cpp'
          Offset:          1447
          Length:          0
          ReplacementText: 'std::move('
        - FilePath:        '/__w/autoware.universe/autoware.universe/planning/behavior_path_planner/src/scene_module/scene_module_bt_node_interface.cpp'
          Offset:          1460
          Length:          0
          ReplacementText: ')'
```

I guess we can send a PR to clang-tidy-pr-comments to fix this behavior.

kenji-miyake, 2022-06-15:
Another result. https://github.com/autowarefoundation/autoware.universe/pull/723

kenji-miyake, 2022-06-21:
I'll try to change the trigger condition of `clang-tidy-pr-comments.yaml` so that we can run it manually.
It's because currently we have too many issues and the check becomes a bit noisy.

Or if somebody is interested, you can implement it instead of me.

kenji-miyake, 2022-08-13:
I think it's resolved by https://github.com/autowarefoundation/autoware-github-actions/pull/172.

closed 2022-08-13 by kenji-miyake