autowarefoundation/autoware_universe #383: Difficulty of estimating velocity at future time t

taikitanaka3, 2022-02-14:
Currently there are behavior and motion planning modules.  Each modules publishes it's own planed velocity (eg. slow down or stop velocity). But this velocity planing is done sequentially from behavior to motion so behavior planning modules doesn't know how motion planing modules plan velocity.

[image]

takayuki5168, 2022-09-22:
@taikitanaka3 Now we calculate the vehicle velocity with motion_velocity_smoother as a library in behavior_velocity_planner. Did this solve this issue?

taikitanaka3, 2022-09-22:
no, but there is no approach with current architecture so i will close this issue.

closed 2022-09-22 by taikitanaka3