autowarefoundation/autoware_universe #490: Refactoring Map Based Prediction

purewater0901, 2022-03-07:
Autoware is currently using map based prediction to predict future trajectories of the object. However, this node has some limitations. 

- It only outputs one possible trajectory .
- It does not calculate confidence correctly compared to its reference paper.
- Predicted yaw angle of the object is not correct.
- The readability of the code is not preferable level.

Therefore, we need to create a new architecture for prediction node to fix these problems.

mitsudome-r, 2023-03-17:
@purewater0901 Do we still need this issue? Or is it solved with https://github.com/autowarefoundation/autoware.universe/pull/491

purewater0901, 2023-03-21:
I think we do not need this issue any more. @mitsudome-r