autowarefoundation/autoware_universe #508: Easier way to find the emergency reason

takayuki5168, 2022-03-11:
Currently we manage emerngey with diagnostics, and we can see its reasons on rqt_robot/runtime_monitor.

However, we do not always run rqt_robot/runtime_monitor, so if the emergency occurrs, we have to run rqt_robot/runtime_monitor and reprocude the emergency again.
IMO, it's better to output emergency reason when it occurrs in standart output log with RCLCPP_WARN/ERR_THROTTLE (WARN for latent fault, and ERROR for single point fault).

It's also very helpful to find the emergency reason of scenario test, which was run on CI since we have standard output log for each scenario test.

takayuki5168, 2022-03-11:
@kenji-miyake What do you think? If you agree, I'll implement this output.

kenji-miyake, 2022-03-11:
@takayuki5168 As a tentative way, I think it is acceptable.
But ideally, we should define the interface for notifying the emergency reason from Autoware to the simulator because the console log isn't a good UI for everyone.
Then the simulator catches the notification and shows it on the result page.

Also, could you show me a screenshot of the result page for reference?

@yukkysaito @mitsudome-r @isamu-takagi @xmfcx Any ideas?

takayuki5168, 2022-03-11:
@kenji-miyake 
> Then the simulator catches the notification and shows it on the result page.
What I mentioned is not only for the scenario test analysis but also for running autoware with simple planning simulator or on the physical vehicle. I don't think it's not good to output the emergency reason on the standard output log for those cases.

Could I talk about how the scenario simulator gets the information of emergency reasons from Autoware in another issue since it's different from what I wanted to discuss :pray:

kenji-miyake, 2022-03-11:
@takayuki5168 I'm sorry, I couldn't understand that well. :cry: 
What you want to do seems to me just adding a hacky interface.

For `simple_planning_simulator`, it's the same.
The console log isn't a good interface because we cannot get it programmatically.
People have to look into the logs and try to find out the reason, it's not efficient or scalable. It's just supplemental information. 

> Could I talk about how the scenario simulator gets the information of emergency reasons from Autoware in another issue

Yes, of course, it's fine.

takayuki5168, 2022-03-11:
I've talked with Miyake-san.
In summary, we first implement just standard output of the emergency reasons, and see whether it works good for debugging from the point of view of development and product service (So far we don't even have a clear requirement). Then we may be able to think about the better idea based on how the current one works.

For the scenario simulator, I'll create another issue as I said.

Any comment would be welcome if you have.

closed 2022-03-18 by takayuki5168