autowarefoundation/autoware_universe #509: Normalizing azimuth angle with wrong offset

yucedagonurcan, 2022-03-11:
Hello,

https://github.com/autowarefoundation/autoware.universe/blob/67459763bd61a57e044132174dd1c403550c1f00/sensing/pointcloud_preprocessor/src/outlier_filter/dual_return_outlier_filter_nodelet.cpp#L180

Doesn't it need to be 360000.f? It looks like there is this error every time we try to normalize the angle.

badai-nguyen, 2022-04-15:
Hello @yucedagonurcan 
I think the line is to make sure the difference of adjacent points in a ring azimuth_diff in range [0.0f  36000.0f].
Could you describe more details about the error have you get?

yucedagonurcan, 2022-04-19:
@badai-nguyen , I am guessing that we need to f_x the difference between [0, 360)_not [0, 36).

badai-nguyen, 2022-04-21:
@yucedagonurcan Hi, I am sorry for misunderstanding your question. 
Because the current LiDAR's azimuth represents angle in hundredths of degree so that it should be 36000.0f.

yucedagonurcan, 2022-04-24:
Okay, it's a fair approach if we know it is `hundredths of degree`

But if we are parsing the sensor data from the Ouster, user tend to calculate the azimuth value as millidegrees which is the format of the incoming data. 

@badai-nguyen  maybe the best approach to make it a parameter rather than fix value? What do you think?

badai-nguyen, 2022-04-26:
@yucedagonurcan yes, I totally agree that it is better to parameterize that. I will fix it.

badai-nguyen, 2022-06-20:
I close this issue because I created an related one here #1127

closed 2022-06-20 by badai-nguyen