autowarefoundation/autoware_universe #535: Max steer angle parameter is defined in each package, not in vehicle_info package

takayuki5168, 2022-03-17:
Since max steer angle parameter is not in the vehicle_info package, this parameter is defined in each packages' yaml file.

takayuki5168, 2022-03-17:
cc @satoshi-ota

takayuki5168, 2022-03-17:
Max jerk, acceleration, velocity are the same. (should be in vehicle_info package)

takayuki5168, 2022-03-17:
@satoshi-ota @kenji-miyake 
Do you agree, and if so, do you have any other parameters to include in the `vehicle_info` package.

kenji-miyake, 2022-03-17:
cc @yukkysaito @mitsudome-r @xmfcx 
I agree that we should manage `max_steer_angle` as a global parameter.
For what else, I have no concrete ideas, but I feel they should be categorized like `shape`, `hard_limit`, `soft_limit`, etc.

Also, as a long-term issue, we should write a design document for global parameters and re-design some packages for Autoware Core.

takayuki5168, 2022-04-17:
@kenji-miyake 
We've already commonized max and min acceleration and jerk. It may be better to write max and min velocity as well here though.
https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_planning_launch/config/scenario_planning/common/common.param.yaml

For the `vehicle_info` package, I don't have any parameters other than `max_steer_angle` so far.

@TakaHoribe 
Could you give us your opinion.

TakaHoribe, 2022-04-19:
I agree to have `max_steer_angle` in the vehicle_info data. There already are some packages defining max_steer value in their local and it's getting messy.

About the velocity, it may be ok to have the max_velocity in the planning common param. In my opinion, we could have two max_velocity: `normal.max_velocity` and `limit.max_velocity`. They are like, "follow the limit if possible" and "vehicle must follow the limit". However the definition of the `normal` and `limit` is a little vague, so it'd be nice to add more detailed description for them when we add a max_velocty.

takayuki5168, 2022-04-20:
I'm thinking about the name of the parameter, `max_steer_angle` or `max_tire_angle`. I'm not sure if `max_steer_angle` is confusing for non-Japanese people.

@xmfcx Could you tell me your opinion.

xmfcx, 2022-04-20:
@takayuki5168 I think `max_steer_angle` is ok. 

[image]

Reference: http://street.umn.edu/VehControl/javahelp/HTML/Definition_of_Vehicle_Heading_and_Steeing_Angle.htm

xmfcx, 2022-04-20:
@TakaHoribe 

> About the velocity, it may be ok to have the max_velocity in the planning common param. In my opinion, we could have two max_velocity: `normal.max_velocity` and `limit.max_velocity`. They are like, "follow the limit if possible" and "vehicle must follow the limit".

What is the difference between these 2? 

Why is a single speed limit not enough, like vehicle is never allowed to go over 50kmph?

The other speed limits are road related, not vehicle.

takayuki5168, 2022-04-21:
@TakaHoribe @kenji-miyake @xmfcx cc @tkimura4 @h-ohta 
We found that `max_steer_angle` is not required for the differential wheeled robot.

So how about the following `vehicle_info.param.yaml`?
Any major/minor comment would be welcome!
```
/**:
  ros__parameters:
    wheel_radius: 0.39
    wheel_width: 0.42
    ...
    vehicle_height: 2.5

    vehicle_model: "differential_model" # select the vehicle model from differential_model or bicycle_model

    differential_model:
        # if there is any parameter for differential model.
    bicycle_model:
        max_steer_angle: 0.70 # [rad]
```

takayuki5168, 2022-04-26:
@TakaHoribe What do you think.

TakaHoribe, 2022-04-26:
> What is the difference between these 2?

@xmfcx In my mind, just one velocity limit is enough.
However, the concept of 2 limits (e.g normal.max_acc and limit.max_acc) is that different accelerations are use for emergency situations (like, a very large acceleration is required to avoid a risk) and non-emergency situations. And I think it is reasonable.

So we can apply the same idea to the velocity limit as well, that is, the two velocity limits for regular and emergency situations. For example, in the case of merging, ego-vehicle may sometimes need to accelerate beyond the velocity limit to avoid a collision to the behind vehicle.

But as you say, the velocity limit should be defined by the road, so just having two velocity limits as I wrote is not a good way.
(I don't think we have to care about this right now though.)

TakaHoribe, 2022-04-26:
@takayuki5168 
The current policy for the adaptation to the differential drive robot is to consider the robot as a virtual Ackerman vehicle. 

In order to support the differential drive robot officially, we need to replace many parts of planning/control/vehicle modules.

Whereas, the drawbacks of the virtual Ackerman vehicle is handling of the non-essential parameters like steering angle and the lack of the pure angular-velocity rotation. This is not that burden compared to the modification of all Ackerman-specific modules.

So I suggest to have max_steer parameter for all types of vehicles (at least for now).

takayuki5168, 2022-04-27:
@TakaHoribe 

> In order to support the differential drive robot officially, we need to replace many parts of planning/control/vehicle modules.

Right. I've been thinking about using (this yaml file)[https://github.com/autowarefoundation/autoware.universe/issues/535#issuecomment-1104791848] with the comment "the differential model is not officially supported. xxx have to be fixed.".

But I found this almost meaningless for now. Now I totally agree with your opinion. Thank you so much.

closed 2022-05-11 by takayuki5168