autowarefoundation/autoware_universe #539: how to use elevation_map_loader in perception?

LLplane, 2022-03-17:
Hello!
What role does "elevation_map_loader" play in perception?
And how to use it?

Think U !

kenji-miyake, 2022-03-17:
@LLplane Ah, I'm sorry. since it was impossible to convert from different repositories, could you ask again?

kenji-miyake, 2022-03-17:
I've updated the settings of issue templates. https://github.com/autowarefoundation/autoware.universe/pull/540

So now you can select a type.
[image]

LLplane, 2022-03-17:
"elevation_map_loader" is new node in aw.universe,and I do not know how to use it. I select Question to ask.

kenji-miyake, 2022-03-17:
Yes, if you open a new question thread, I will `@mention` the author, and he would explain it.

LLplane, 2022-03-17:
Sorry, I do not know how to open a new question, can not find the page like the above picture.

kenji-miyake, 2022-03-17:
Here! Just click the `New issue` button.

[image]

LLplane, 2022-03-17:
ooo,think u !