autowarefoundation/autoware_universe #549: Cannot get Velodyne Lidar UDP packet in docker container ( --host network mode ) by tier4 Velodyne driver

xczhanjun, 2022-03-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The perception module cannot  get Velodyne UDP packet in docker container ( by --host network mode ) by tier4 Velodyne driver, but the https://github.com/ros-drivers/velodyne/tree/ros2 drive can.

### Expected behavior

The perception module cannot  get Velodyne UDP packet in docker ( b--host network mode ) by tier4 Velodyne driver, but the https://github.com/ros-drivers/velodyne/tree/ros2 drive can.

### Actual behavior

The official https://github.com/ros-drivers/velodyne/tree/ros2 drive can work.

### Steps to reproduce

just run ros2 launch velodyne_dirvers in latest -amd64-prebuilt(20220315) docker

kenji-miyake, 2022-03-20:
Please write more clear steps to reproduce. For example, what (all and exact) commands you executed, what data you used, etc.

xczhanjun, 2022-03-20:
> Please write more clear steps to reproduce. For example, what (all and exact) commands you executed, what data you used, etc.

xczhanjun, 2022-03-22:
@kenji-miyake hi, today I tested the latest code using docker --net=host mode again, the velodyne_driver can get the udp packet from host network for a while, but it cannot work after a reboot. I will send you a msg when I figured out the root cause.

kenji-miyake, 2022-03-22:
I understand, thank you for your investigation! :smile:

mitsudome-r, 2022-06-21:
@xczhanjun 
> hi, today I tested the latest code using docker --net=host mode again, the velodyne_driver can get the udp packet from host network for a while, but it cannot work after a reboot. I will send you a msg when I figured out the root cause.

Were you able to find the root cause of the issue?

xmfcx, 2022-07-26:
@xczhanjun are there any updates on this issue? Are you still having this issue?

mitsudome-r, 2022-10-25:
Since there are no response, I will close this issue. @xczhanjun Feel free to reopen as necessary.

closed 2022-10-25 by mitsudome-r