autowarefoundation/autoware_universe #551: Implement sensor drivers for ROS 2

BonoloAWF, 2022-03-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Implement sensor drivers for ROS 2

### Purpose

The Bus ODD is being developed in ROS 2 which requires porting sensor drivers where required

### Possible approaches

Find open source sensor drivers or implement the required drivers

### Definition of done

- [ ] List of sensors on the bus 
- [ ] List of missing ROS 2 drivers 
- [ ] Development of missing drivers

mitsudome-r, 2022-03-23:
Yang shared https://github.com/Kitware/LidarView, which might help us design a unified sensor driver for different LiDARs

TakumiKozaka-T4, 2022-03-24:
@BonoloAWF 
Could you remove my name from assignees?
I'm not concerned with this issue.

> [TakumiKozaka-T4](https://github.com/TakumiKozaka-T4) pushed a commit to TakumiKozaka-T4/autoware.universe that referenced this issue [22 hours ago](https://github.com/autowarefoundation/autoware.universe/issues/551#ref-commit-bd4111f)

I didn't push anything. I only updated my forked repository.

senagolcuk, 2022-04-04:
| Sensor Type | Sensor Model      | Sensor Driver                                          |
|-------------|-------------------|--------------------------------------------------------------|
| Radar       | Delphi ESR        |                                                              |
| Lidar       | Ouster OS1-64     | [OS1-64 ROS2 Driver](https://github.com/ros-drivers/ros2_ouster_drivers/tree/ros2) |
| GNSS-INS    | Trimble BX992-INS |                                                              |
| Camera      | Onsemi AC-AR0231  |                                                              |

senagolcuk, 2022-04-04:
| Sensor Type | Sensor Model      | Sensor Driver                                          |
|-------------|-------------------|--------------------------------------------------------------|
| Radar       | Delphi ESR        |   Missing                                                           |
| Lidar       | Ouster OS1-64     | [OS1-64 ROS2 Driver](https://github.com/ros-drivers/ros2_ouster_drivers/tree/ros2) |
| GNSS-INS    | Trimble BX992-INS |                     Missing                                             |
| Camera      | Onsemi AC-AR0231  |                  Missing                                                |

mitsudome-r, 2022-05-10:
@mitsudome-r will check with ITRI about ROS2 drivers for their platform

BonoloAWF, 2022-05-23:
@hueirutseng what is the status update regarding upgrading ROS 2 drivers for the Bus ODD ITRI platform? Are there any missing drivers?

jason914, 2022-05-25:
|Sensor Type       | Sensor Model            |Sensor driver          |Status        |                
| ----------------- |------------------------|---------------------|------------|
|Radar                  |Delphi ESR                 |Missing                  |Work in progress|
|Lidar                    |Ouster OS1-64          |[OS1-64-ROS2 driver](https://github.com/ros-drivers/ros2_ouster_drivers/tree/ros2)|Tested OK |
|GNSS-INS            |Trimble BX992-INS    |Missing                  |Work in progess|
|Camera                |Onsemi AC-AR0231 | Private code(Please contact [Neousys](https://www.neousys-tech.com/en/) for source code)|Tested OK|

mitsudome-r, 2022-06-07:
@jason914 Thanks for posting the table.
Do you have any updates + estimated timeline on Radar and GNSS-INS sensor driver?

mikechan0731, 2022-06-08:
Sensor Type | Sensor Model | Sensor driver | Status
-- | -- | -- | --
Radar | Delphi ESR | Missing | WIP, Estimated test date is the end of June.
Lidar | Ouster OS1-64 | OS1-64-ROS2 driver | Tested OK
GNSS-INS | Trimble BX992-INS | Missing | WIP, Estimated test date is the end of August.
Camera | Onsemi AC-AR0231 | Private code(Please contact Neousys for source code) | Tested OK

mitsudome-r, 2022-06-21:
Expected message for GNSS is explained here: https://github.com/autowarefoundation/autoware_msgs/pull/24

mitsudome-r, 2022-07-05:
Once we have all drivers available, we can merge the information with the list created in the hardware WG and post it in autoware website.

xmfcx, 2022-07-19:
@senagolcuk could you update here with the sensor drivers used in the ISUZU bus project?

@meliketanrikulu can you share the estimated timeline for the open sourcing of the applanix driver?

xmfcx, 2022-07-26:
GNSS-INS driver will be published here: https://github.com/autowarefoundation/applanix

mitsudome-r, 2022-07-26:
@senagolcuk 
Could you tell me why your sensor driver status for lidar is "Work in progress"?
If you have the link to the driver that you are using, could you share it as well?

senagolcuk, 2022-07-27:
@mitsudome-r 

I'm updating now.

meliketanrikulu, 2022-08-01:
> Applanix POS LVX GNSS-INS driver will be published here: https://github.com/autowarefoundation/applanix

Added applanix POS LVX GNSS-INS driver.

sglee-morai, 2022-08-02:
I have a question. Is it planned that [OS1-64-ROS2 driver](https://github.com/ros-drivers/ros2_ouster_drivers/tree/ros2) will be included in the Autoware.Universe like Velodyne drivers are? I see OS1-64 in the comment above, but cannot find a PR regarding that.

mitsudome-r, 2022-08-02:
@mikechan0731 Could you update [this comment](https://github.com/autowarefoundation/autoware.universe/issues/551#issuecomment-1149385309) to add the link to the drivers that you are using?

senagolcuk, 2022-08-02:
@meliketanrikulu thanks for Applanix Driver. I'm updating the table for the new driver.

jason914, 2022-08-11:
|Sensor Type       | Sensor Model            |Sensor Driver Status        |                
| ----------------- |------------------------|-----------|
|Radar                  |Delphi ESR                 |WIP, Estimated completion date is the end of August|
|Lidar                    |Ouster OS1-64          |[ros2-ouster-driver](https://github.com/ros-drivers/ros2_ouster_drivers) |
|GNSS-INS            |Trimble BX992-INS    |WIP, Estimated completion date is the end of August|
|Camera                |Onsemi AC-AR0231 | Private code, Camera manufacturer does not provide code.|

xmfcx, 2022-08-16:
@senagolcuk could you update the list with the https://github.com/autowarefoundation/lucid_vision_driver driver?

senagolcuk, 2022-08-16:
@xmfcx I updated it.

mitsudome-r, 2022-08-30:
Regarding ITRI's comment: https://github.com/autowarefoundation/autoware.universe/issues/551#issuecomment-1211529554
GNSS-INS driver is ready for testing, but the code is private now. They need internal review to find if they can share the code.

xmfcx, 2022-09-06:
Since we've gathered possible sensor drivers, we can close this issue.

xmfcx, 2022-09-06:
https://github.com/autowarefoundation/autoware-projects/wiki/Bus-ODD-Project related.

closed 2022-09-06 by xmfcx