autowarefoundation/autoware_universe #562: Collect raw sensor data to test sensor drivers

mitsudome-r, 2022-03-22:


Sharrrrk, 2022-04-18:
@BonoloAWF please assign this task to me.

Sharrrrk, 2022-04-19:
> https://drive.google.com/file/d/13oypA5UdlW1XIqw6l5C3qf0I4sFaZ1Nz/view?usp=sharing
> 
> > Copyright 2022 AutoCore, Inc.
> > Uploder: @Sharrrrk
> > Record Date: 14th.Apr.2022
> 
> ### Description:
> 
> This folder contains pcap files and rosbag files from Ouster OS1-64 Lidar. The pcap file and rosbag file is recored in the same time with slight difference in duration.
> 
> Senario1: multiple static trucks on road side. -- trucks.pcap -- rosbag2_2022_04_14-trucks
> 
> Senario1: Pedstrain moving around vehicle with static vehicle on road side. -- ped_vehicle.pcap -- rosbag2_2022_04_14-ped_vehicle

Using the driver below:
https://github.com/ros-drivers/ros2_ouster_drivers

Sharrrrk, 2022-04-27:
Data has been provided

closed 2022-04-27 by Sharrrrk