autowarefoundation/autoware_universe #565: Evaluate performance of perception pipeline

mitsudome-r, 2022-03-22:


kaancolak, 2022-03-28:
In the Autoware.Auto, we have 2D and 3D detection benchmark tools for the KITTI dataset. We can test our 2D/3D detections by integrating these tools into Autoware.Universe.
 
 In addition, I think, we need to evaluate the entire perception stack of Autoware.Universe from sensor output(Point Cloud/Camera Frame) to object tracking output, because all perception modules work together. Such as; deterministic and deep learning-based 3D bounding boxes are based merged or tracking results are merged with detected objects. For object tracking, the KITTI dataset has only 2D bounding box-based evaluation. For this reason, we are planning to implement Waymo dataset for 3D detection & tracking evaluation.

kaancolak, 2022-04-11:
I shared high-level architecture for the planned perception benchmark pipeline. If you have any comments feel free to share them with us. 

[image]

**Limitations when benchmarking with the Waymo dataset:** 

- We don't have PCD/Vector Map information from the Waymo dataset. The current perception pipeline can run with/or without PCD Map.  
- In our codebase, a cylindrical prism is used when assigning shapes to pedestrians, we need 3D bounding boxes for pedestrians when evaluating with the Waymo dataset.

**Current situation:**
- I created a ROS interface that contains data conversion functionalities for the Waymo dataset.
- I fed the data from the Waymo dataset to the lidar-only Autoware.Universe pipeline.

[image]

WJaworskiRobotec, 2022-05-23:
@kaancolak I'm wondering if using Waymo Open Dataset Toolkit for evaluation is a good idea. I understand that the best solution would be to make it possible to use the same metric calculation software for Waymo, Kitti ( + other open datasets), as well as with synthetic data generated in the simulators. In this case, as Autoware is based on ROS2, it would be perfect to have metric calculation based on ROS topics directly, optionally on rosbags (like we develop in this [issue](https://github.com/autowarefoundation/autoware-projects/issues/11)) What is your opinion ? I'm not really familiar with Waymo Open Dataset toolkit, so I might misunderstand something.

WJaworskiRobotec, 2022-05-25:
Thanks a lot for the explanation. Pipeline that you created is perfect for comparing Autoware results with others, as well as it will be very easy to just connect metric calculation node that is created in the task I mentioned directly to the same topics that you connect the "data converter" converting to the Waymo format. Look perfect to me and once we have proper metrics we will connect it with your code.

kaancolak, 2022-05-27:
@WJaworskiRobotec Instead of working on the different benchmarking pipelines, I think we can extend the metric calculation node. It's a more generic evaluator that contains the entire stack(evaluator, planning, control, detection, etc.) that makes sense. I would like to implement the perception(2D/3D perception) part on your metric calculation nodes, base algorithm will be very similar to this tool but must be in the same code format as your nodes.

WJaworskiRobotec, 2022-05-27:
@kaancolak Sounds great. @djargot is currently working on the last node that we wanted to create as an example and it is related to perception (segmentation algorithm evaluation). Once it's done we will assign you as a reviewer, and you can continue with adding your nodes for 2D/3D Object Detection.

kaancolak, 2022-06-20:
This PR waiting for review.

xmfcx, 2022-09-20:
@kaancolak can you update the current status of this issue?

kaancolak, 2022-09-20:
I have made some updates on the code base. Currently, it's waiting for review.

mitsudome-r, 2022-10-25:
@kaancolak If you have any evaluation results, please

closed 2022-10-25 by mitsudome-r