autowarefoundation/autoware_universe #566: Evaluate current planning pipeline

mitsudome-r, 2022-03-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Investigate performance of planning pipeline.

### Purpose

Evaluate current planning pipeline and confirm that it has enough features for Bus ODD.

### Possible approaches

Test planning pipeline against scenarios created by ODD WG.

### Definition of done

- [ ] Get the list of scenario files from ODD WG.
- [ ] Run scenario tests manually against the planning pipeline
- [ ] List failed tests and investigate the cause of failure.
- [ ] Create issues for missing features or bugs found from failed tests.

mehmetdogru, 2022-04-05:
Some of the .yaml scenario files are created by ODD wg but testing them with autoware.universe is not possible at the moment because of the current state of autoware.universe.

sglee-morai, 2022-05-23:
For the scenario _UC-PB-005-0001-Hsinchu.yaml_, which I originally created, I see the note is written like below,

> * Vehicle doesn't stay on the stop line
> * Vehicle sometimes can't make the right turn because of the narrow drivable area and gets stuck on the right turn. Fails with timeout"

The problem is that there is NO explicit stop line in the initial lane where the ego-vehicle is positioned initially. So please let me know if the scenario file itself should be modified.

mehmetdogru, 2022-05-24:
@sglee-morai 

The reason why I wrote this was that in ODD Design Document, initial case is defined as "EGO is stopping at the stop line" but the vehicle is not stopped on the stop line and it should be written :) You are right about that if the vehicle is already stopped before the turn, there is no difference between whether there is a stop line or not. It would matter tho if the initial vehicle speed was not zero, however no difference in this case.

There are no stop lines available on the lanelet2_map which would satisfy this _(UC-PB-005-0001)_ scenario design so there is nothing to be done on your end regarding the scenario content. Therefore I edited `Related Part` column for the scenario file as **lanelet2_map**. 

Thank you for the attention and the fine work.

sglee-morai, 2022-05-31:
@mehmetdogru Thanks a lot Mehmet for your confirmation! 

> @sglee-morai
>
> The reason why I wrote this was that in ODD Design Document, initial case is defined as "EGO is stopping at the stop line" but the vehicle is not stopped on the stop line and it should [...]

mitsudome-r, 2022-06-06:
For UC-PB001-003, I have made an update to the scenario file. https://gitlab.com/autowarefoundation/operational-design-domains/-/merge_requests/26

xmfcx, 2022-07-19:
For the initial testing, I think we can call this complete with the results in: https://github.com/autowarefoundation/autoware.universe/issues/566#issuecomment-1102729980

If we need to evaluate it again, we can recreate another issue.

closed 2022-07-19 by xmfcx