autowarefoundation/autoware_universe #567: Evaluate control modules in Universe

mitsudome-r, 2022-03-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Investigate performance of control pipeline in universe.

### Purpose

Evaluate current control pipeline and confirm that it has enough features for Bus ODD.

### Possible approaches

Create benchmark tools to test control modules (e.g. calculate offsets from trajectory) and evaluate control moduels

### Definition of done

- [ ] Design a benchmark tool to evaluate control modules (also check with Sim WG)
- [ ] Run benchmark tools against control modules
- [ ] Create issues for missing features or bugs found from the results.

brkay54, 2022-04-04:
@mitsudome-r There is a package called control_performance_analysis in Control. I am confused, is purpose of this issue develop this package? Or, this package is not going to be used anymore so we need to develop a new package?

Mentioned package: https://github.com/autowarefoundation/autoware.universe/tree/main/control/control_performance_analysis

mitsudome-r, 2022-04-05:
@TakaHoribe Could you explain what control_performance_analysis is used for, and what kind of metrics are used to generate the result.

boyali, 2022-04-07:
This package computes;
tracking error outputs:
- lateral error ey
- heading error eyaw

The tracking errors are computed by projecting current vehicle location on the current trajectory. In addition, the generated control vector u us used to compute an quadratic energy like value similar to we use in the MPC computations; uRu^T. We use the same kind of energy for the tracking’s errors; xPx^T. 

The cost matrices R and P can simply be taken as unity matrices. However in the package we use an R and P matrices computed from stability LMI equations for a kinematic feedback controller.

In addition in the package we compute an approximate curvature for each waypoint on the trajectory.

brkay54, 2022-04-08:
We decided to modify the ` control_performance_analysis` package. I will add new functionalities to this package. Current design is here:

[image]
 
I will create PR for this design. If you have any comment, please feel free to share with us.

brkay54, 2022-06-14:
Example usage of control performance analysis tool: https://github.com/orgs/autowarefoundation/discussions/412#discussioncomment-2947841

mitsudome-r, 2022-08-02:
@brkay54 Thank you for running the test.

Discussion from ASWG:
* @brkay54 to try more tests in different map with different path (e.g., with more curves)
* Also, providing screenshots of the tests might help us imagine what is causing the larger error in `mpc_follower`

brkay54, 2022-08-09:
There are same updates about testing.
There were two blockers for me.

- Firstly, I realized that heading and lateral error were increasing so much in straight road. So I investigated that and I found that `control_performance_analysis` tool sometimes choose wrong nearest trajectory point. It was fixed with this [PR](https://github.com/autowarefoundation/autoware.universe/pull/1522).

- Secondly, I tried run the evaluation tool in different maps and with long time range. But PlotJuggler's buffer size is not enough to record all data. I opened issue for this. 

For now, I am currently run this test on kashiwanoha_map because of second blocker (I can not collect the all data for long time period). I collected data in kashiwanoha_map, after some processes, I am going to share here.

brkay54, 2022-08-10:
Both MPC and pure_pursuit were tested in three different environment:

1. sample_vehicle + kashiwanoha_map
2. sample_vehicle + gebze_map
3. isuzu_vehicle + gebze_map

## Driving status

* Test - 1 sample_vehicle + kashiwanoha_map

| Series               | PP Min    | MPC Min   | PP Max   | MPC Max  | PP Average | MPC Average |
|----------------------|-----------|-----------|----------|----------|------------|-------------|
| lateral_acceleration | -0.403416 | -0.547744 | 0.481827 | 0.618049 | 0.020227   | 0.017232    |
| lateral_jerk         | -1.983906 | -4.221178 | 2.424854 | 2.450203 | -0.013859  | -0.002141   |

* Test - 2 sample_vehicle + gebze_map

| Series               | PP Min    | MPC Min   | PP Max   | MPC Max  | PP Average | MPC Average |
|----------------------|-----------|-----------|----------|----------|------------|-------------|
| lateral_acceleration | -0.502603 | -0.517804 | 0.18027  | 0.215238 | -0.040143  | -0.040282   |
| lateral_jerk         | -1.276513 | -1.565028 | 0.913313 | 1.330014 | -0.000146  | -0.000129   |

* Test - 3 isuzu_vehicle + gebze_map

| Series               | PP Min    | MPC Min   | PP Max   | MPC Max  | PP Average | MPC Average |
|----------------------|-----------|-----------|----------|----------|------------|-------------|
| lateral_acceleration | -0.779351 | -0.82164  | 0.271746 | 0.308302 | -0.062338  | -0.062065   |
| lateral_jerk         | -1.985067 | -2.340564 | 1.413154 | 1.961581 | -0.00036   | -0.000516   |

* MPC has higher lateral acceleration and jerk peak value than pure_pursuit

## Performance variables

* Test - 1 sample_vehicle + kashiwanoha_map

| Series                     | PP Min    | MPC Min   | PP Max   | MPC Max  | PP Average | MPC Average |
|----------------------------|-----------|-----------|----------|----------|------------|-------------|
| heading_error              | -0.443751 | -0.579112 | 0.402318 | 0.421394 | 0.002471   | -0.007      |
| heading_error_velocity     | -0.191558 | -0.203922 | 0.248491 | 0.376283 | 0.001939   | -0.000965   |
| lateral_error              | -0.381273 | -0.22516  | 0.439361 | 0.439361 | 0.009929   | 0.046452    |
| lateral_error_acceleration | -0.424266 | -0.586141 | 0.674285 | 1.010908 | 0.011949   | 0.008286    |
| lateral_error_velocity     | -0.887937 | -0.905332 | 0.882841 | 0.917043 | 0.006962   | 0.007918    |

* Test - 2 sample_vehicle + gebze_map

| Series                     | PP Min    | MPC Min   | PP Max   | MPC Max  | PP Average | MPC Average |
|----------------------------|-----------|-----------|----------|----------|------------|-------------|
| heading_error              | -0.355129 | -0.290287 | 0.152418 | 0.063587 | -0.005227  | -0.003886   |
| heading_error_velocity     | -0.102993 | -0.092189 | 0.111945 | 0.126462 | -0.000953  | -6.2E-05    |
| lateral_error              | -0.280567 | -0.289781 | 0.220632 | 0.222301 | -0.014059  | -0.00631    |
| lateral_error_acceleration | -0.318318 | -0.268501 | 0.319817 | 0.354164 | -0.002302  | -0.002464   |
| lateral_error_velocity     | -0.819191 | -1.08276  | 1.149198 | 0.27054  | -0.014403  | -0.015096   |

* Test - 3 isuzu_vehicle + gebze_map

| Series                     | PP Min    | MPC Min   | PP Max   | MPC Max  | PP Average | MPC Average |
|----------------------------|-----------|-----------|----------|----------|------------|-------------|
| heading_error              | -0.604754 | -0.188215 | 0.175775 | 0.04927  | -0.00514   | -0.004605   |
| heading_error_velocity     | -0.338885 | -0.397858 | 0.109483 | 0.12697  | -0.022434  | -0.021345   |
| lateral_error              | -0.30969  | -0.351939 | 0.231677 | 0.233544 | -0.014029  | -0.012137   |
| lateral_error_acceleration | -0.915756 | -1.127259 | 0.489062 | 0.571718 | -0.078175  | -0.076123   |
| lateral_error_velocity     | -0.945196 | -1.023388 | 0.740805 | 0.217162 | -0.014718  | -0.018148   |

## Performance variables total error (RMS)

* Test - 1 sample_vehicle + kashiwanoha_map

| Series                                       | PP          | MPC         |
|----------------------------------------------|-------------|-------------|
| rms_heading_error                            | 555.11382   | 589.214804  |
[... 23 lines cut ...]
| rms_lateral_acceleration_error               | 2811.365921 | 3094.668373 |
| rms_lateral_error                            | 1249.054939 | 1289.547955 |
| rms_lateral_velocity_error                   | 1103.374622 | 890.818035  |
| rms_tracking_curvature_discontinuity_ability | 15.135825   | 15.161382   |

## Results Plotted (Lateral Error - Heading Error - XY Pose of vehicle)

* Test - 1 sample_vehicle + kashiwanoha_map

[image]

* Test - 2 sample_vehicle + gebze_map

[image]

## To visualize the results

* Download the data: [alltests.tar.gz](https://github.com/autowarefoundation/autoware.universe/files/9300623/alltests.tar.gz)
* Open PlotJuggler and import the data
* Now, you can see the series at left side of window. You can plot the data.
* To plot vehicle position for debugging, Select two curves x and y from vehicle kinematic state (keeping the CTRL key pressed) and Drag & Drop them using the RIGHT Mouse button.

## My opinion

There is no big performance difference I think. Pure Pursuit has better performance in kashiwanoha_map than MPC. However, MPC has better performance in Gebze map. I think Pure Pursuit can better handle the sharp turns than MPC but generally MPC has better performance.

brkay54, 2022-08-24:
@mitsudome-r I added instruction documentation in PR. We can close the issue now.

mitsudome-r, 2022-08-30:
@brkay54 Thanks for all your work! I'm closing this issue.

closed 2022-08-30 by mitsudome-r