autowarefoundation/autoware_universe #583: LaserscanToOccupancyGrid should be in perception package

taikitanaka3, 2022-03-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

it should in perception package(porting miss)

### Expected behavior

move package to perception

### Actual behavior

move package to perception and launch as perception package

### Steps to reproduce

see perception launch and check if there is laserscan to occupncy grid is launched https://github.com/tier4/autoware_launch/blob/a232bfa812da37e254c773cf4bafb8b0a62ca74b/perception_launch/launch/perception.launch.xml#L47

mitsudome-r, 2023-03-17:
@taikitanaka3 Can we close this issue? https://github.com/autowarefoundation/autoware.universe/pull/1923 seems to be merged already.

taikitanaka3, 2023-03-17:
@mitsudome-r yes

closed 2023-03-17 by taikitanaka3