autowarefoundation/autoware_universe #589: Perception packages support CUDA 11.4 and TensorRT 8.2.2

yukke42, 2022-03-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Perception packages are ported from https://github.com/tier4/AutowareArchitectureProposal.iv and some of them use CUDA 11.1 and TensorRT 7.2.1, but https://github.com/autowarefoundation/autoware requires CUDA 11.4 and TensorRT 8.2.2.
- traffic_light_ssd_fine_detector
- traffic_light_classifier
- lidar_centerpoint
- tensorrt_yolo
- lidar_apollo_instance_segmentation

### Purpose

The perception packages work on the https://github.com/autowarefoundation/autoware environement.

### Possible approaches

Refer to https://github.com/tier4/tensorrt_common to switch the TensorRT version.

### Definition of done

The packages are conformed to work both versions.

yukke42, 2022-04-06:
I confirmed lidar_centerpoint works. https://github.com/autowarefoundation/autoware.universe/issues/643#issuecomment-1090090877

wep21, 2022-04-07:
apollo
[image]

wep21, 2022-04-07:
yolo
[image]

wep21, 2022-04-07:
traffic light
[image]

yukke42, 2022-04-07:
Thank you! :clap:

yukke42, 2022-04-07:
We checked for the packages to work, so close this issue.

closed 2022-04-07 by yukke42