autowarefoundation/autoware_universe #592: Create calibration tool for sensors

mitsudome-r, 2022-03-28:


xmfcx, 2022-03-29:
https://github.com/ethz-asl/kalibr

https://github.com/ApolloAuto/apollo/blob/master/docs/quickstart/apollo_2_0_sensor_calibration_guide.md

@xmfcx fix

mitsudome-r, 2022-03-29:
During the call, we have decided to split the tickets into a smaller task. @mitsudome-r will do work on it.

xmfcx, 2022-04-12:
@xmfcx provide details for issues 
- https://github.com/autowarefoundation/autoware.universe/issues/677
- https://github.com/autowarefoundation/autoware.universe/issues/679
- https://github.com/autowarefoundation/autoware.universe/issues/678

xmfcx, 2022-07-19:
Already being discussed in:
- https://github.com/autowarefoundation/autoware.universe/issues/679
- https://github.com/autowarefoundation/autoware.universe/issues/677

closed 2022-07-19 by xmfcx