autowarefoundation/autoware_universe #610: Add YOLO Custom Dataset Handler

goktugyildirim, 2022-03-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Before training the YOLO object detector, we need to collect, manage, and combine the training and testing samples by being aware of their classes and label counts, etc. In order to facilitate the process, we need the YOLO label integration scripts for celebrated datasets consisting of related classes in autonomous driving in order to combine them and also the script for analyzing the whole samples coming from different datasets. Also, we need a script to send queries to the combined celebrated dataset database to generate our target dataset easily.

### Purpose

The purpose of this feature is to make generating a custom dataset for the YOLO object detector easy using combined celebrated datasets such as KITTI, Waymo Open Dataset, etc.

### Possible approaches

Firstly, universal 2D object detection classes and important classes can be defined. Then running YOLO label integration scripts to provide a sample database with original labels. After providing YOLO label files for each sample, training huge YOLO nets like YOLOV4-p6 or YOLOV4-p7 for each important class element using maybe a million labels. We can do pseudo labeling over the samples to complete the missing class. For instance, KITI Dataset doesn't consist of a traffic sign class and we can train a traffic light detector using LISA Traffic Sign Dataset to do pseudo labeling over the KITTI dataset to increase our custom dataset size. Also, the handler script can take the specific queries to generate a custom dataset.

### Definition of done

Open to discussing

goktugyildirim, 2022-03-30:
It can be named Custom Dataset Composer, it makes sense. In Fact, pseudo labels will be improving our search algorithm on the datasets. For instance, you want to train a custom detector with the target classes of "pedestrian" and "vehicle", and you have two different datasets having every single class.

dataset1: pedestrians (original labels)
dataset2: vehicle (original labels)

In this case, you cannot take a sample from dataset1 if it contains a "vehicle" class. After "vehicle" pseudo labeling, you can find out which sample contains "vehicle" in dataset1 because the one class detector that has the high test scores can easily find out if the target sample contains a related class or not.  You can take samples from dataset1 not contains "vehicle". In sum, my approach is based on sample identification instead of Semi-Automatic Labeling. Also, one class detectors can improve the mAP of the YOLO model in the case of having too little sample from a class, even if the test scores of some classes are decreased.

xmfcx, 2022-04-04:
> You cannot combine the two datasets if they don't have common target classes.

I think we should take the datasets as they are and train only with the datasets with matching contents. If not it cannot be used. 

The datasets come with their list of labels that make clear what they contain. Anything else should be assumed non-labeled.

mitsudome-r, 2022-05-24:
@xmfcx Will do the creation of repo and update of repos this week.

xmfcx, 2022-05-31:
@goktugyildirim I've created the repository here: https://github.com/autowarefoundation/custom_dataset_composer and gave you maintainer access on it.

goktugyildirim, 2022-05-31:
> @goktugyildirim I've created the repository here: https://github.com/autowarefoundation/custom_dataset_composer and gave you maintainer access on it.

Thanks, @Zeysthingz is planning to take this issue on herself.

mitsudome-r, 2022-06-21:
@Zeysthingz Do you have any updates on this issue?

Zeysthingz, 2022-06-21:
@mitsudome-r Yes I do. I will push updates  to repository but I try to understand how to do it in Autoware Universe. Probably I will update the repository in these two days.

Zeysthingz, 2022-07-06:
@xmfcx Fatih can you assign available reviewers for pull request?

closed 2022-07-06 by xmfcx