autowarefoundation/autoware_universe #611: Need to speed bump behavior module

beyzanurkaya, 2022-03-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The behavior velocity planner has several modules in it. like "blind_spot", "crosswalk", "stop_line", "traffic_light" etc. But a module is needed for the speed bump. First of all, in this module, the bump zone on the map should be understood. And the vehicle must decrease its normal speed according to previous road conditions at a safe distance from the speed bump. After completing a pass over the bump, the vehicle should continue its normal course. 

### Purpose

The purpose of this task is to create a speed bump module. so that the behavior of the ego vehicle on a bumpy road is defined. In this way, the ego vehicle can drive safer.

### Possible approaches

Write code for speed bump module

### Definition of done

- [ ] Should map data be treated and identified as a bump region
- [ ] In the module, the behavior of the Ego vehicle in the bump zone should be defined.
- [ ] Test the module with the planning simulator.

xmfcx, 2022-04-05:
@beyzanurkaya please add the PR's to this issue.

mitsudome-r, 2022-09-27:
@mehmetdogru @beyzanurkaya Please update the PR once you are finished with field validation.