autowarefoundation/autoware_universe #619: tier4_sensing_launch now depends on common_sensor_launch

h-ohta, 2022-04-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

`tier4_sensing_launch` now depends on `common_sensor_launch`.

https://github.com/autowarefoundation/autoware.universe/blob/95414c5d3ff944016dab9376d82d7fdceeb10e4e/launch/tier4_sensing_launch/package.xml#L12

### Purpose

I think `common_sensor_launch` is not always needed when developping autoware.

### Possible approaches

I propose to delete this. 

### Definition of done

`common_sensor_launch` is deleted from package.xml in `tier4_sensing_launch`.

h-ohta, 2022-04-04:
@kenji-miyake Do you have time?

kenji-miyake, 2022-04-04:
@h-ohta Hmm? What do you mean? I think we can merge it if you send a PR.

h-ohta, 2022-04-04:
@kenji-miyake OK. I'll make PR.

PR #626 (2022-04-04): fix: delete common_sensor_launch from exec_depend
## Description

- Resolves #619.
- Delete `common_sensor_launch` from exec_depend since this package is not refered from `tier4_sensing_launch`

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [x] The PR follows the [pull request guidelines].

## Post-review  [... cut]
closed 2022-04-04 by h-ohta