autowarefoundation/autoware_universe #622: dummy_perception_publisher publishes objects with acceleration

takayuki5168, 2022-04-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently dummy_perception_publisher can publish the obstacle with the constant velocity. To realize more flexible use case, I'm thinking to add acceleration/deceleration functions to dummy_perception_publisher. This function will be helpful for obstacle stop/slow down (including adaptive_cruise_controller).

### Purpose

> To realize more flexible use case, I'm thinking to add acceleration/deceleration functions to dummy_perception_publisher. This function will be helpful for obstacle stop/slow down (including adaptive_cruise_controller).

### Possible approaches

The current message to represent initial pose/twist of objects is `InitialState`.
https://github.com/autowarefoundation/autoware.universe/blob/main/simulator/dummy_perception_publisher/msg/InitialState.msg

Instead of `InitialState`, how about `Motion` or `State` message as follows.
```
geometry_msgs/PoseWithCovariance pose_covariance
geometry_msgs/TwistWithCovariance twist_covariance
double acceleration
```

### Definition of done

Objects' acceleration can be changed with dummy_perception_publisher.

takayuki5168, 2022-04-01:
@yukkysaito

yukkysaito, 2022-04-04:
@takayuki5168 
I agree with adding accel.

>how about Motion or State message as follows.

This was a little confusing.
Are you saying that there is a type called Motion, State? If so, please provide a link.

takayuki5168, 2022-04-04:
@yukkysaito 
> Are you saying that there is a type called Motion, State? If so, please provide a link.

Sorry to be confused. There is no existing messages. I just named Motion or State, whose content is as follows.
```geometry_msgs/PoseWithCovariance initial_pose_covariance
geometry_msgs/TwistWithCovariance initial_twist_covariance
double acceleration
```

takayuki5168, 2022-04-04:
I was thinking that in dummy perception publisher, in order to represent the case where the obstacle is running and will stop (= initial twist linear x is positive and acceleration is negative), the updated velocity (acceleration is added every cycle) will be made non-negative.
But sometimes we may want to make obstacle running backward, which has a conflict with a rule that the update velocity is non-negative.

If we have max_velocity and min_velocity variable in a new message, which I named Motion or State, the conflict can be resolved, but I feel the number of variables is too many.
Do you have any good idea.

takayuki5168, 2022-04-05:
max_velocity and min_velocity can be in config file of dummy_perception_publisher. Those values cannot be changed every obstacles though.

yukkysaito, 2022-04-05:
I would prefer the following
```
geometry_msgs/PoseWithCovariance initial_pose_covariance
geometry_msgs/TwistWithCovariance initial_twist_covariance
geometry_msgs/AccelWithCovariance initial_accel_covariance
```
- The dummy perception may consider in the future.
- Pose and twist are also cov for the above reason.

>max_velocity and min_velocity can be in config file of dummy_perception_publisher. Those values cannot be changed every obstacles though.

I agree with you.

takayuki5168, 2022-09-22:
Fixed by https://github.com/autowarefoundation/autoware.universe/pull/1853

closed 2022-09-22 by takayuki5168