autowarefoundation/autoware_universe #628: Add TVM inference capability

ambroise-arm, 2022-04-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Port the `tvm_utility` and `neural_networks` packages from Autoware.Auto.

### Purpose

Provide TVM inference capability in autoware.universe.

### Possible approaches

Either:

1. Port the packages as they are (downloading compiled artifacts from an S3 bucket).
2. Or, because the current way of getting the artifacts is not perfect, change it to use git-lfs.

This requires some discussion. I've written on what the situation is here: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/merge_requests/1258#note_899430670 and I will create a proper GitHub discussion when I'm sure that the change (if wanted) will only be done on the GitHub side.

### Definition of done

- `tvm_utility` and `neural_networks` are merged
- a docker setup exists that enables running the inference on a GPU backend (vulkan); if not with the default setup, then a dedicated one must be offered to the user

mitsudome-r, 2022-05-24:
related issue https://github.com/autowarefoundation/autoware.universe/issues/908

ambroise-arm, 2022-06-08:
Will need to address https://github.com/autowarefoundation/autoware/issues/128 for the vulkan backend to work.

xmfcx, 2022-08-16:
https://github.com/autowarefoundation/autoware.universe/pull/1181 is waiting to be reviewed.

ambroise-arm, 2022-09-30:
done: all merged

closed 2022-09-30 by ambroise-arm