autowarefoundation/autoware_universe #630: Some errors are occurred in the tutorial

yukkysaito, 2022-04-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

There are several error outputs in the tutorial.

### Expected behavior

There are No error.

### Actual behavior

There are several error outputs in the tutorial.

### Steps to reproduce

Planning simulation
`ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/Downloads/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/

Rosbag replay simulation
`ros2 launch autoware_launch logging_simulator.launch.xml map_path:=$HOME/Downloads/sample-map-rosbag vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/

yukkysaito, 2022-04-05:
@ito-san Can you investigate the error regarding hdd_monitor?

xmfcx, 2022-04-12:
Are https://github.com/autowarefoundation/autoware.universe/issues/641 and https://github.com/autowarefoundation/autoware.universe/issues/643 are part of this issue?

xmfcx, 2022-07-19:
Since the tasks need to be re-done, we'll close the issue and if errors occur again, this issue needs to be re-created.

yukkysaito, 2022-11-10:
cc @ito-san 
Now, I want to reopen this issue because we get some errors when plannign simulator is launched.
[image]

ito-san, 2022-12-01:
@yukkysaito Please consider to close this issue if you don't have any problem with it after the fixes.

xmfcx, 2023-01-10:
@yukkysaito is this issue still present?

mitsudome-r, 2023-03-17:
I'm closing the issue since there are no updates for long time.
We can reopen this issue or create new one if we find issue with the tutorial.

closed 2023-03-17 by mitsudome-r