autowarefoundation/autoware_universe #641: The planning simulator does not work as per the tutorial.

yukkysaito, 2022-04-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The planning simulator does not work as per the tutorial.
It seems to the pull over function of the behavior path planner does not publish.
[image]
[image]

### Expected behavior

https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/

### Actual behavior

The planning simulator does not work as per the tutorial.

### Steps to reproduce

https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/

rej55, 2022-04-07:
It seems to be happen because the status of scene modules is not initialized.
https://github.com/autowarefoundation/autoware.universe/blob/7ee98f4768870e0f33688922c68143faa3acd29f/planning/behavior_path_planner/include/behavior_path_planner/scene_module/scene_module_interface.hpp#L202

The initial status should be `BT::NodeStatus::IDLE`.
I'll fix it.

yukkysaito, 2022-04-07:
@rej55 Thank you!

yukkysaito, 2022-04-07:
New issues : 
- Poor trajectory follower performance and no parking state.
[image]

- If a goal is set quickly at startup, it cannot move.
[image]

mitsudome-r, 2022-06-21:
@yukkysaito Do we still have issues with the tutorial? I believe they are working with the newest code.

xmfcx, 2022-07-19:
We'll close this issue and it can be re-opened if it occurs again.

closed 2022-07-19 by xmfcx