autowarefoundation/autoware_universe #643: The rosbag replay simulator does not work as per the tutorial. #641

yukkysaito, 2022-04-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The rosbag replay simulator does not work as per the tutorial.
- centerpoint doesn't publish the topic
- traffic_light_ssd_fine_detector doesn't work

[image]

### Expected behavior

https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/

### Actual behavior

The rosbag replay simulator does not work as per the tutorial.
- centerpoint doesn't publish the topic
- traffic_light_ssd_fine_detector doesn't work

### Steps to reproduce

https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/

wep21, 2022-04-06:
Could you check if gdown successfully download the weight files?
```
❯ ls install/traffic_light_ssd_fine_detector/share/traffic_light_ssd_fine_detector/data/
mb2-ssd-lite-tlr.onnx  voc_labels_tl.txt
```
```
❯ ls install/lidar_centerpoint/share/lidar_centerpoint/data/
pts_backbone_neck_head_aip_x2.onnx     pts_backbone_neck_head_default.onnx  pts_voxel_encoder_default.engine
pts_backbone_neck_head_default.engine  pts_voxel_encoder_aip_x2.onnx        pts_voxel_encoder_default.onnx
```

yukkysaito, 2022-04-06:
@wep21 thank you
yes, too many downloads are failing in gdown.
That was the reason why traffic_light_ssd_fine_detector was not working.
But the centerpoint weights file is downloaded manually. (engine file is created at initial runtime)
```
yukky@yukky:~/workspace/tutorial/autoware$ ls src/universe/autoware/perception/lidar_centerpoint/data/
pts_backbone_neck_head_aip_x2.onnx  pts_backbone_neck_head_default.onnx  pts_voxel_encoder_aip_x2.onnx  pts_voxel_encoder_default.onnx
```

yukkysaito, 2022-04-06:
I manually down loaded the traffic light and the error disappeared.

yukkysaito, 2022-04-06:
lidar_apollo_instance_segmentation_node does not work, too :cry:

yukke42, 2022-04-06:
This error is possibly caused by TensorRT version. https://github.com/autowarefoundation/autoware.universe/issues/589

wep21, 2022-04-06:
@yukkysaito Could you check the tensorrt version?
```
❯ apt-cache policy libnvinfer-dev
libnvinfer-dev:
  インストールされているバージョン: 8.2.4-1+cuda11.4
  候補:               8.2.4-1+cuda11.4
```

yukke42, 2022-04-06:
Using docker lidar_centerpoint can be built and run successfully, but the subscribed topic name to it is weird.
[image]

yukke42, 2022-04-06:
I launch autoware without perception and launch lidar_centerpoint in other terminal, the detected objects are visualized on rviz.
[image]

yukke42, 2022-04-15:
@yukkysaito 

Was this issue solved by [this PR](https://github.com/autowarefoundation/autoware.universe/pull/645#event-6381035116)?

yukkysaito, 2022-06-10:
@yukke42 Yes! Thank you

closed 2022-06-10 by yukkysaito