autowarefoundation/autoware_universe #649: Sometimes trajectory_follower's test fails with ARM64

kenji-miyake, 2022-04-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

See https://github.com/autowarefoundation/autoware.universe/runs/5860423885?check_suite_focus=true.

### Expected behavior

CI passes.

### Actual behavior

CI failed.

### Steps to reproduce

Run the `build-and-test-self-hosted` workflow several times.

### Versions

ARM64

### Additional context

Previously, it passed if I re-run the workflow. So I guess it's just flaky.

kenji-miyake, 2022-04-07:
@TakaHoribe @taikitanaka3 Do you know any causes?

maxime-clem, 2022-04-08:
Previously flaky tests were sometimes caused by interference between tests that use the same topics running in parallel.
I have opened a draft PR https://github.com/autowarefoundation/autoware.universe/pull/661 to switch the tests of the `trajectory_follower_nodes` to run in isolation but it is hard to verify if this solves the issue.

kenji-miyake, 2022-04-08:
@esteve cc @mitsudome-r @xmfcx Are you familiar with this issue? I remember you've worked on this kind of task in Autoware.Auto.
If so, could you give me your thoughts? :pray:

closed 2022-04-12 by maxime-clem