autowarefoundation/autoware_universe #653: fail-safe-state image is not found

h-ohta, 2022-04-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The following link is not found, https://github.com/autowarefoundation/autoware.universe/blob/c341d4c4e4a79dd0b1f2259653a4f6cc3f7e4c50/system/emergency_handler/README.md?plain=1#L11.
Does anyone know acorrect url ?

### Expected behavior

We can see the fail-safe-state image.

### Actual behavior

We cannot see the fail-safe-state image.

### Steps to reproduce

1. access https://tier4.github.io/autoware.proj/tree/main/design/apis/image/fail-safe-state.drawio.svg

kenji-miyake, 2022-04-07:
https://github.com/autowarefoundation/autoware.universe/pull/644

h-ohta, 2022-04-07:
I'm sorry I overlooked...

h-ohta, 2022-04-12:
also #635 
closed

closed 2022-04-12 by h-ohta