autowarefoundation/autoware_universe #662: free space planner may collide with front

kosuke55, 2022-04-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I think that the following lines shorten the forward length of the vehicle and may cause a collision.
``` 
 for (double x = back; x <= front; x += costmap_.info.resolution) {
    for (double y = right; y <= left; y += costmap_.info.resolution) {
```
https://github.com/autowarefoundation/autoware.universe/blob/main/planning/freespace_planning_algorithms/src/abstract_algorithm.cpp#L135-L136

It's better to guarantee that `x(y)` exceeds `font(left)` like
``` 
 for (double x = back; x <= front + costmap_.info.resolution; x += costmap_.info.resolution) {
    for (double y = right; y <= left + costmap_.info.resolution; y += costmap_.info.resolution) {
```

### Expected behavior

The front of the vehicle does not collide with obstacles

### Actual behavior

Sorry, I haven't reproduced it but the front of the vehicle may collide with obstacles.

### Steps to reproduce

Sorry, I haven't reproduced it but the front of the vehicle may collide with obstacles.

kosuke55, 2022-04-10:
cc @TakaHoribe @yukkysaito

mitsudome-r, 2022-05-10:
@kosuke55 @TakaHoribe Is anyone from TIER IV going to fix the issue?

kosuke55, 2022-05-10:
@HiroIshida  @TakaHoribe 
If you can check if this is a mistake, I will send fix PR.

HiroIshida, 2022-05-11:
@kosuke55 
Sorry for late reply. I will check today.

HiroIshida, 2022-05-12:
Thank you, I think you are right.

kosuke55, 2022-07-17:
@TakaHoribe @rej55 
Not urgent, but if you have time, please check https://github.com/autowarefoundation/autoware.universe/pull/985 :bow:

closed 2022-08-03 by kosuke55