autowarefoundation/autoware_universe #673: Create visualizer for system_error_monitor

h-ohta, 2022-04-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The diagnostics_aggregator is visualized in [rqt_robot_monitor](http://wiki.ros.org/rqt_robot_monitor), and DiagnosticsArray messages are also visualized in [rqt_runtime_monitor](http://wiki.ros.org/rqt_runtime_monitor).

However, we don't have any visualizer for `system_error_monitor` now.
For example, when we fall into emergency, we have to echo some topics from `system_error_monitor` with terminal and analyse them to resolve.

### Purpose

The purpose of this task is to create some visualization ways to check system_error_monitor status with GUI more easily.
I believe this function can able to make our developers to resolve the problem like emergency more quickly. 

### Possible approaches

Various approaches are considered
- create rqt plugin same as rqt_robot_monitor
- create rviz plugin
- create other GUI approaches

### Definition of done

- [ ] created some visualization ways
- [ ] We can check the status and the problem by using them

kenji-miyake, 2022-04-12:
@h-ohta Have you tried `ros2 run rqt_runtime_monitor rqt_runtime_monitor --ros-args -r /diagnostics:=/diagnostics_err`?

h-ohta, 2022-04-12:
Yes, I tried it.
I confirmed when a diag are staled, [Single Point Fault] are added to it.
However, I can't confirm emergency and emergency_holding status from it.

closed 2026-03-24 by mitsudome-r