autowarefoundation/autoware_universe #677: Investigate available open source tools for calibration.

xmfcx, 2022-04-12:


xmfcx, 2022-07-19:
Can you add them here: https://github.com/autowarefoundation/autoware-documentation/blob/main/docs/how-to-guides/sensor-calibration.md ?

xmfcx, 2022-07-19:
Previous issue: https://github.com/autowarefoundation/autoware.universe/issues/677

raminterpl, 2022-08-02:
1.) SensorsCalibration toolbox v0.1

SensorsCalibration is a simple calibration toolbox and open source project, mainly used for sensor calibration in autonomous driving.
https://github.com/PJLab-ADG/SensorsCalibration

**Lidar2camera:**
https://github.com/PJLab-ADG/SensorsCalibration/blob/master/lidar2camera/README.md
https://github.com/PJLab-ADG/SensorsCalibration/tree/master/lidar2camera

**Lidar2Lidar:**

https://github.com/PJLab-ADG/SensorsCalibration/blob/master/lidar2lidar/README.md

2.) Mathworks calibration tool: 
**Lidar Camera Calibrator**
Interactively estimate rigid transformation between lidar sensor and camera

https://in.mathworks.com/help/lidar/ug/lidar-camera-calibration.html
https://in.mathworks.com/help/lidar/ref/lidarcameracalibrator-app.html

mitsudome-r, 2022-08-02:
@raminterpl Thank you for invetigation.
Could you create PR to https://github.com/autowarefoundation/autoware-documentation/blob/main/docs/how-to-guides/calibrating-your-sensors.md to share the results in the documentation?
Once that is done, we can close this issue.

mitsudome-r, 2022-08-16:
Looks like there is already a PR so we can close the issue.
https://github.com/autowarefoundation/autoware-documentation/pull/173#pullrequestreview-1073609330

closed 2022-08-16 by mitsudome-r