autowarefoundation/autoware_universe #701: Change longitudinal controller from rate to period control

Sharrrrk, 2022-04-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Change the control type of longitudinal controller from rate to period.

### Purpose

Currently in trajectory_follower_nodes package, lateral controller is using parameter _ctrl_period_ to define the period, while logitudinal controoller is using _control_rate_ to define the rate. But on real vehicles, period is used to defined how long a control command should be sent. 
For example, the _control_rate_ of logitudinal controller is defined as 30 by default, which leads to the period of the logitudinal control rate to be 33.33333 ms, which is not a possible period for real vehicle control interface. On the other hands, in code, rate is transferred to period to setup a timer callback, which is not necessary.
The purpose of this task is to change the control type of longitudinal controller, so that control period is used to defind how long the longitudinal control command should be sent, so that it consistant with lateral controller and real vehicle.

### Possible approaches

Replace the parameter _ctrl_period_ to _longitudinal_ctrl_period_, and update the code and doc accordingly.

### Definition of done

- [ ] Code update.
- [ ] Document update.
- [ ] Tested.

TakaHoribe, 2022-04-15:
Thank you for your issue! I totally agree with this approach.

JianKangEgon, 2022-04-15:
One thing to mention is that the duplicated name in parameter configuration file may cause conflict, so it's better to add prefix / suffix if necessary. Although, the namespace is supposed to be useful, but actually, from my own experience, it does not work at all......

TakaHoribe, 2022-04-15:
@JianKangEgon Good point. In ros2, every parameter is basically defined under each node. Thus we have to call the parameter with the node name like `ros2 param get /trajectory_follower/lateral_controller control_rate`. This is exactly the mechanism introduced in ros2 to prevent the parameter conflicts you are concerned about. Does this solve your concern?

Also, at the code level, the parameter is stored in yaml file defined for each node.

JianKangEgon, 2022-04-19:
@TakaHoribe Thanks for the reply. Well, I understand your point and believe this kind of mechanism of ros2 functionality as well. However, I did meet such a confusing problem during my actual migration of autoware new Architecture proposal  about two or three months ago.  Here, I'd like to reproduce this issue in details as following:

- ubuntu 20.04
- ROS2 Galactic
- Autoware version: https://github.com/tier4/AutowareArchitectureProposal.proj

To realize transformation of the autoware control command to vehicle command, we have one vehicle driving-by-wire package named byd_dbw, its configuration file is shown as following:
[image]

We can see, there is one parameter named "control_rate" set to 100.0, to configure the frequency of vehicle command. And this package is launched in vehicle module. 

In comparison, the control_rate of velocity controller is defined in velocity_controller.launch.xml as "30.0" by default. And this node is launched in control module. 

When we run the autoware_launch file, we can see the interesting result:
[image]

The topic hz of longitudinal controller is then set to 100. Although the parameters are defined in totally different modules and launched in different groups, there is still such an unpleasant problem occurred. Our solution is to rename the duplicated parameters.

Currently, we are working on migration of the autoware.universe to our real vehicle. It's much more reasonable to define each single parameter in yaml file rather directly in lauch file. After several tests, there does not exist conflict any more when duplicated names in different modules.

TakaHoribe, 2022-04-25:
@JianKangEgon Thanks for sharing. This looks strange to me and I don't know why this is happening.

In this case, is the `longiturinal_controller/control_rate` parameter set to 100?

JianKangEgon, 2022-04-25:
> @JianKangEgon Thanks for sharing. This looks strange to me and I don't know why this is happening.
> 
> In this case, is the `longiturinal_controller/control_rate` parameter set to 100?

@TakaHoribe Well, I was shocked at this problem as well and took a long time to figure it out. The longiturinal_controller/control_rate is always set to 30.0 by default. The planning simulator works always well because there is no any other control_rate defined in the launch file.

The first time I found this issue was about 3 months ago during our road test on site. The autoware_launch did not success to run the longitudinal controller all the time, and printed the "mismatched data type" on the screen. (The control_rate of byd_dbw was set to 100 at the very beginning, so there was the parameter declaration conflict between "int" and "double"). And we checked the launch file (longiturinal_controller/control_rate defined directly in it) again and again to confirm that this param was set correctly, until we found that there existed another control_rate even in different module in different launch file. 

Just amazing, for now, this issue seems to disappear after several attempts under the current mechanism to load parameter config. 

Thanks again for your reply and I'd love to share these unexpectable cases.

Sharrrrk, 2022-04-27:
Merged.

closed 2022-04-27 by Sharrrrk