autowarefoundation/autoware_universe #703: Add another crop_box_filter node

beginningfan, 2022-04-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Add another crop_box_filter node to crop LiDAR reflections from ego vehicle body.
#437 added vehicle crop box filter, but it is missing in current code of autoware.universe.

### Purpose

At present, there is only one crop_box_filter node , but actually we need two crop_box_filter nodes for real cases. For example, one crop_box_filter to to remove reflections from far away objects or tunnel surface/upper tree branches, the other crop_box_filter crop reflations of the ego vehicle, which will cause planning module to decelerate. So multiple crop_box_filter is required to achieve this.

### Possible approaches

Add another crop_box_filter node here [link](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_perception_launch/launch/obstacle_segmentation/ground_segmentation/ground_segmentation.launch.py), and also need to add the parameters here [link](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_perception_launch/config/obstacle_segmentation/ground_segmentation/ground_segmentation.param.yaml)

### Definition of done

- [ ] Code complete.
- [ ] Tested.

xmfcx, 2022-04-19:
https://github.com/autowarefoundation/autoware-documentation/pull/79#discussion_r853202192

We should add some node like this to filter the points within the polygon https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/tree/master/src/perception/filters/polygon_remover

xmfcx, 2022-04-26:
Vehicle params:  https://github.com/autowarefoundation/autoware.universe/blob/main/vehicle/vehicle_info_util/config/vehicle_info.param.yaml

mitsudome-r, 2022-06-21:
@beginningfan 
Now that https://github.com/autowarefoundation/autoware.universe/pull/924 is merged can we close this issue?

closed 2022-06-21 by beginningfan