autowarefoundation/autoware_universe #706: Wrong path planned by avoidance module in a special case

shulanbushangshu, 2022-04-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When the ego vehicle encounters obstacles near discontinuous adjacent lanes (see picture below), part of the path planned by avoidance module of behavior_path_planner exceeds the driveable area.
This triggers the corresponding protection function of obstacle_avoidance_planner to stop the ego vehicle.
However, after the obstacle disappears, the vehicle will not withdraw from the obstacle avoidance behavior because the offset path has been calculated, thus causing the vehicle to be stuck in place.
[image]

### Expected behavior

Planned path shall be within the driveable area, not entering undefined area in vector map.
Ego vehicle shall continue to drive after the obstacle disappears in the scene above.

### Actual behavior

Part of the planned path exceeds the driveable area and enters the undefined area of map.
After the obstacle disappears, the ego vehicle stands still.

### Steps to reproduce

1. Start autoware planning simulator using default
2. Set a obstacle near discontinuous adjacent lanes (like picture below)
3. Engage
4. Delete the obstacle

### Versions

- OS: Ubuntun 20.04
- ROS2:  Galactic
- Autoware: Autoware.universe:main

### Possible causes

behavior_path_planner only check the max shift point in planned path, which is this case, is within the driveable area. This check does not apply on other points of the planned path. But obstacle_avoidance_planner will check all the points of the planned path.

yukkysaito, 2022-04-15:
Can you provide a map and rosbag?
cc @takayuki5168

TakaHoribe, 2022-04-15:
Thank you for your issue! 
We have recognized this issue before (please see the [design document - unimplemented parts - cancel avoidance when target disappears](https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_path_planner/behavior_path_planner_avoidance-design/#future-extensions-unimplemented-parts)). 
And we are in the process of improvement.

There should be more items to be fixed, as you say, the behavior_path should be inside the drivable area. The avoidance still has an issue in this kind of edge place (where the lane boundary changes along the path).

closed 2022-06-21 by shulanbushangshu