autowarefoundation/autoware_universe #710: PathFootprint/Trajectory Footprint's size is not the same as the vehicle size.

takayuki5168, 2022-04-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Currently, path footprint/trajectory footprint's size is a constant value, which should be the vehicle's size from the vehicle_info package.
https://github.com/autowarefoundation/autoware.universe/blob/1e829e2948222e8b9c60a2af3bf3008ffbe2ec6d/common/tier4_planning_rviz_plugin/src/path_footprint/display.cpp#L35-L40

### Expected behavior

Path footprint/trajectory footprint's size will be the vehicle's size.

### Actual behavior

Path footprint/trajectory footprint's size is a constant value.

### Steps to reproduce

visualize path footprint or trajectory footprint plugin

takayuki5168, 2022-04-15:
@kenji-miyake I'm not sure how to use vehicle_info parameters (length, width, ...) for path footprint/trajectory footprint, which is an rviz plugin for visualization.
Could you give me a hint to fix this issue.

kenji-miyake, 2022-04-16:
I guess we can get the parameters if we declare it. Will give it a try.

```
~/autoware main
❯ ps aux | grep rviz
kenji    1317153 13.6  1.7 1455300 564052 pts/5  Sl+  16:41   2:43 /opt/ros/galactic/lib/rviz2/rviz2 -d /home/kenji/autoware/install/autoware_launch/share/autoware_launch/rviz/autoware.rviz -s /home/kenji/autoware/install/autoware_launch/share/autoware_launch/rviz/image/autoware.png --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_2f1hcudt
kenji    1384063  0.0  0.0  10452  2692 pts/2    S+   17:01   0:00 grep --color=auto rviz

~/autoware main
❯ cat /tmp/launch_params_2f1hcudt
/**:
  ros__parameters:
    front_overhang: 1.0
    left_overhang: 0.1
    rear_overhang: 1.03
    right_overhang: 0.1
    use_sim_time: false
    vehicle_height: 2.5
    wheel_base: 2.74
    wheel_radius: 0.39
    wheel_tread: 1.63
    wheel_width: 0.42
```

kenji-miyake, 2022-04-16:
@takayuki5168 Yes, it was possible.

```
❯ ros2 param dump --print /rviz2
/rviz2:
  ros__parameters:
    front_overhang: 1.0
    left_overhang: 0.1
    qos_overrides:
      /parameter_events:
        publisher:
          depth: 1000
          durability: volatile
          history: keep_last
          reliability: reliable
    rear_overhang: 1.03
    right_overhang: 0.1
    use_sim_time: false
    vehicle_height: 2.5
    wheel_base: 2.74
    wheel_radius: 0.39
    wheel_tread: 1.63
    wheel_width: 0.42
```

kenji-miyake, 2022-04-16:
I've tested this With extremely expanded vehicle_info parameters.

Left: Launched RViz using `ros2 launch autoware_launch` (= has global params)
Right: Launched RViz using `rviz2 -d`

[image]

PR #712 (2022-04-23): feat(tier4_planning_rviz_plugins): add vehicle_info to Footprint Displays
## Description

Resolves #710 

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [x] The PR follows the [pull request guidelines].

## Post-review checklist for the PR author

The PR author **must** check the checkboxes below before mer [... cut]
closed 2022-04-23 by takayuki5168