autowarefoundation/autoware_universe #711: Goal pose is not aligned to the map

takayuki5168, 2022-04-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When the goal pose is given to Autoware, usually by "2D Goal Pose" rviz plugin, mission_planner recognized that the goal pose's z is 0, and publishes the goal pose marker whose z is 0.
This may cause some misleading to any calculation using the goal pose or visualization.

### Expected behavior

For "2D Goal Pose", since it does not set a z value as the name suggests, mission planner should calculate the proper z value based on the lanelet so that the goal pose will be aligned to the lanelet.

### Actual behavior

The goal pose's z is 0.

### Steps to reproduce

Just set the goal pose with "2D Goal Pose" rviz plugin.

takayuki5168, 2022-04-16:
@yukkysaito @mitsudome-r What do you think.
If the goal pose given to mission_planner may contain the z value in the future, we have to discuss more.

takayuki5168, 2022-04-18:
cc @TakaHoribe @kosuke55

takayuki5168, 2022-04-20:
@yukkysaito @mitsudome-r kindly ping.

xmfcx, 2022-06-14:
We can republish the 2D Goal Pose from the mission planner with correct Z coordinates.

@angry-crab would like to work on this.

closed 2022-07-04 by xmfcx