autowarefoundation/autoware_universe #742: Need to add time stamp for control command in emergency handler

taikitanaka3, 2022-04-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

no time stamp for lat/lon control command
https://github.com/autowarefoundation/autoware.universe/blob/26a252e446a030ecac6fc13449a04048086dbba1/system/emergency_handler/src/emergency_handler/emergency_handler_core.cpp#L307

### Expected behavior

visible time stamp

### Actual behavior

no time stamp

### Steps to reproduce

awf: autoware.universe main
tier4: autoware_launch tier4/universe

### Versions

awf: main

taikitanaka3, 2022-04-21:
@kenji-miyake 
Is there any diagnostic error for this?

kenji-miyake, 2022-04-21:
@taikitanaka3 I'm sorry, could you explain it a bit more clearly? :pray:

taikitanaka3, 2022-04-21:
~Currently there is no emergency control command lateral command time stamp and longitudinal command stamp.
Does this cause any problem for product? I want to make the exact priority to this issue.
So I want to know if there is any fail safe for this ...~
But it looks like there is no problem for now. I am sorry to bother you.

kenji-miyake, 2022-04-21:
Hmm? Is it resolved?

> I am sorry to bother you.

No problem! Let's discuss it whenever you find anything strange. :smile:

taikitanaka3, 2022-04-21:
I am now asked wakabayashi-san to solve this issue. but as far as I now there is no package refer to this time stamp I set low priority label.

shmpwk, 2022-04-24:
Close by #747

closed 2022-04-24 by shmpwk